Edzelf/ESP32Radio-V2 #124: Is it possible to add 1 new mqqt command.

SijDijk, 2023-09-09:
I use multiple touch screens and PCs (multiple rooms) connected via mqqt but if the esp32 radio is already playing I would like to send a command so that all information is sent again just like when the esp is started (Stream name, Track information, Volume , Preset and possibly IP address).

Now it is the case that when the screen is switched on later, only Track information appears and it is actually not my intention to reset the ESP when a new screen is registered.

I have been trying to understand the software for quite some time now, but I cannot find where I can possibly add a new command.

with kind regards
SijDij

SijDijk, 2023-12-12:
Does anyone else know how to add an extra mqtt command for the above problem?

Edzelf, 2023-12-14:
I added the command "mqttrefresh".  This will trigger all MQTT items to refresh.
No time to test it, so please check it out.

SijDijk, 2024-01-21:
It took a while because the USB connector came loose from the ESP32, but it has now been repaired.

I tried "mqttrefresh" and it works exactly as I wanted it to.
Thank you very much.