Edzelf/ESP32Radio-V2 #125: A few notes

Pako2, 2023-09-10:
First of all I have to thank you for the great work you have done on this project. I was looking for what I should start from when constructing an Internet radio and your project turned out to be perfect for it.
I was originally based on a version that was about 5 months old. I made quite a few changes in the meantime, so it was quite difficult for me to upgrade to the new version.
But in any case, I had planned to solve the work with the SD card in a better way, so the new version came in handy.
However, I ran into a few bugs:
**1) Mute-unmute**
This is a good change, if it concerns control via a web interface or perhaps via a serial interface. But there is no button on my remote that can be used for the "unmute" command. So I made a small modification to the code in the SCANIR routine:
```
      val = nvsgetstr ( mykey ) ;                           // Get the contents
      if ((val=="mute") && muteflag)
      {
          val = "unmute";
      }
```
**2) Callback function for handle_mp3list**
There are even 2 bugs in this feature that cause the first and last track to be missing from the mp3 track selection dropdown and one track is always random.
First you need to change
```
  if ( index == 0 )                                   // First call for this page?
  {
    i = 0 ; 
```
thus:
```
  if ( index == 0 )                                   // First call for this page?
  {
    i = 1 ; 
```
This is because a few lines later, **_path = getSDFileName ( i++ )_** is used;
As we know, the meaning of i++ is that i is incremented by one, and then it returns the **old** value of i !
This means that the first time **_getSDFileName ( i++ )_** is called with i=0 it results in a random track being returned.
A zero value of i has another undesirable consequence. The part of the code where the separator is inserted is skipped (and therefore the first track is missing).

Next, the **_if ( i >= SD_filecount )_** code needs to be changed to **_if ( i > SD_filecount )_**, because otherwise the last track is missing.

I made many other adjustments - for example, when selecting a track using the encoder, I can scroll through the list backwards (including jumping to the last track if I get "before" the first track).

Pako2, 2023-09-13:
I ran into another bug. This manifests itself in a certain situation when controlling via the web interface.
How you can achieve the error behavior:
1. Select a track using the drop-down menu
2. Use the NEXT button on the same page - it won't work as expected.
The problem is that when selecting a track using the drop-down menu, the **SD_curindex** variable is not set.
Then nothing that uses this variable can work - for example, selecting the next track with the NEXT button.
I fixed it in the following way:
1. in the javascript of the mp3play.html page, I modified the **_trackreq( )_** function:
```
   function trackreq ( presctrl )
   {
    if ( presctrl.value != "-1" )
    {
      httpGet ( "track=" + encodeURIComponent ( tracks[presctrl.value] ), false ) ;
    }  
  }
```
  
thus:
```
   function trackreq ( presctrl )
   {
    if ( presctrl.value != "-1" )
    {
      httpGet ( "trackinx=" + encodeURIComponent ( document.getElementById("seltrack").value ), false ) ;
    }  
  }
```
This resulted in the tracklist index being used instead of the track name (when calling the GET method).

2. To make it work, I added a new argument "trackinx" to the **_analyzeCmd()_** function in main.cpp:
```
   else if ( argument == "trackinx" ) // MP3 track request?
   {
     getSDFileName ( ivalue );            // Select file by index
     myQueueSend ( sdqueue, &startcmd ) ; // Signal SDfuncs()
   }
```

Now it works as expected.

Pako2, 2023-09-16:
I have another note.
I had planned to use this "radio" as an audiobook player, but I found that in its current form it is unusable for this purpose.
The audiobooks I use are in the form of several mp3 files. The names can be something like part01.mp3, part02.mp3, etc.
When playing, the individual files must follow each other in alphabetical order. And here is the problem. Before copying the audiobook to the SD card (on the PC), the files are sorted. However, they are no longer sorted on the SD card. I tried to solve it and now I have a version that works as expected.
The tracks (in the tracklist.dat file) are sorted alphabetically (however, not globally - always only within one directory).
If you are interested, I can of course provide you with my solution.

Pako2, 2023-09-30:
I still have to add one correction to the adjustments I made.
Although the following code is working
`httpGet ( "trackinx=" + encodeURIComponent ( document.getElementById("seltrack").value ), false ) ;`
that's pretty clumsy. A better version is this:
`httpGet ( "trackinx=" + encodeURIComponent ( presctrl.value ), false ) ;`

lubiezunia, 2025-02-27:
Thanks for mute/unmute modification!