Edzelf/ESP32Radio-V2 #127: Errors Arduino IDE

kondorcl, 2023-10-05:
Hi Ed, please help me. I updated from old version(with VS1053 work fine) Mon, 28 Jun 2021 12:40:00 GMT to V2 Fri, 16 Jun 2023 07:30:00 GMT both version compiled with
ArduinoIDE 1.8.12 and ESP32 2.0.11
V2 i have errors:
Starting ESP32-radio running on CPU 1 at 240 MHz.
E (10491) gpio: gpio_set_level(226): GPIO output gpio_num error
E (10492) gpio: gpio_set_level(226): GPIO output gpio_num error
E (10502) gpio: gpio_set_level(226): GPIO output gpio_num error
E (10502) gpio: gpio_set_level(226): GPIO output gpio_num error
E (14294) gpio: gpio_set_level(226): GPIO output gpio_num error
E (14294) gpio: gpio_set_level(226): GPIO output gpio_num error
E (14294) gpio: gpio_set_level(226): GPIO output gpio_num error
E (14300) gpio: gpio_set_level(226): GPIO output gpio_num error

Best Regards,
ERamirez

Edzelf, 2023-10-09:
The errors are caused by not initialized GPIO numbers.  Probably caused by missing definitions in the config page of the web-interface.  Start by defining DUMMYTFT in config.h if you have not set-up the configuration for your display yet.
I have made some modifications to reduce the number of error messages.

kondorcl, 2023-10-09:
Hi Ed, thank very much for ur answer. I had #define DEC_VS1053  and #define BLUETFT in config.h. I had to remove LittleFS from my libraries, because at startup I had error messages regarding to LittleFS. Now works fine.