Edzelf/ESP32Radio-V2 #129: Problems to build V2 with platformia

JocB1, 2023-10-15:
Hi, 
earlier I had already built the V1 with arduino. Now I would like to build the V2 with Platformio, because I use this already for my 3D Marlin firmware builds. 

Here is what I have done: In "Visual Studio Code" installed the plugin "Espressif IDF v1.6.5". Then opened the file "ESP32-Radio.code-workspace". 

Under Platformio I started the build process in the tree under esp32 -> General -> Build. I also tried before the build "full clean". 
So far all config files are in original state as downloaded here. 

Unfortunately I get the following error message: 

 *  Task wird im Ordner ESP32Radio-V2-main ausgeführt: C:\Users\....\.platformio\penv\Scripts\platformio.exe run --environment esp32 

Processing esp32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)

-------------------------------------------------------------------------------------------------------------------------------

MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package

 *  Der Terminalprozess "C:\Users\.....\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32'" wurde mit folgendem Exitcode beendet: 1. 

What exactly would I need to do with this package.json do? 
Is there a step by step guide for building with Platformio under Visual-Studio-Code? 

Best regards,
Jochen

Edzelf, 2023-10-16:
Espressif IDF is not used for this project. 
The "Espressif ESP32" platform should be installed automatically, check your installed platforms.
After you have installed VScode, the only step is to install the "Platform IDE" extension.

JocB1, 2023-10-16:
Yes, I have installed expressiv only as a plugin when there were without error messages.

I have now uninstalled everything and reinstalled VSCode and Platformio. The original problem is gone.

However, now I get the following error message:

 

Compiling .pio\build\esp32\lib908\Adafruit ST7735 and ST7789 Library\Adafruit_ST7789.cpp.o
Compiling .pio\build\esp32\lib908\Adafruit ST7735 and ST7789 Library\Adafruit_ST77xx.cpp.o

 

src/main.cpp: In function 'void playtask(void*)':
src/main.cpp:4389:51: error: void value not ignored as it ought to be
                            ini_block.shutdownx_pin);
                                                   ^
Compiling .pio\build\esp32\lib2da\AsyncTCP\AsyncTCP.cpp.o
Compiling .pio\build\esp32\lib5dd\FS\FS.cpp.o
Compiling .pio\build\esp32\lib5dd\FS\vfs_api.cpp.o
*** [.pio\build\esp32\src\main.cpp.o] Error 1
================================================= [FAILED] Took 11.88 seconds =================================================

 

 

 

 

Gesendet: Montag, 16. Oktober 2023 um 10:53 Uhr

Edzelf, 2023-10-18:
Yes, my fault.  Please download the new versions of VS1053.cpp and VS1053.h from the lib/codecs directory.

JocB1, 2023-10-20:
Hi, 
thank you very much! With this changes ist works well. 

Best regards, 
Jochen