Edzelf/ESP32Radio-V2 #130: SD-Card activation /  GPIO config

JocB1, 2023-10-20:
Hi, 

I use the V1 with SD-Card support without remarkable problems. Now I flashed the V2 from 2023-10-18 and I can´t get the SD-Card support running. 
I know, SD-Card support is still experimental, but it worked on my hardware with V1. 

In conifig.h I´ve commented SDCARD in. 

So far, at least the radio itself is running. 

When I now add my GPIO (pin_sd_cs = 21) to the Web-based conifg editor an save the config, ESP32Radio-V2 reboots in an endless loop. 
Fortunately, the web interface is always available at the right time for a short time, so I was able to undo the config change.  

Best regards, 
Jochen

Szvzs, 2024-03-01:
I have this message : ***ERROR*** A stack overflow in task SDtask has been detected. Tried connecting through VS1053 (red) SD card slot, also TFT(red) SD card slot with no success. I can leave uncommented the SD in the config file, the radio works, as soon as i connect the cable SD_CS an uncomment the parameter in the webserver it will start resetting endlessly. Did i forget anything ? Where to connect pin_sd_detect ? Thanks again!

Edzelf, 2024-03-05:
The SD card is experimental.  You need quite some free RAM to make it work.  I know that not all cards work on the ESP32.  The SD-detect is available on some SD-modules.   It is a micro switch that is activated when the card is inserted.
Try some different cards with only a few MP3 files on it.