Edzelf/ESP32Radio-V2 #132: Choppy playback, chunks 0 frequently. High latency,

MrSirric, 2023-12-01:
My playback is pretty choppy. Seems to be network related. When connecting to a station directly on my computer or phone its fine. However on the esp32 I'm getting some 0 chunks listed from time to time. Im very new to this platform and application. Would i expect network wise the connections to be of similar quality? 

This issue is on all of my 5 esp32 devices, All radio stations. Bitrate ranged from ~200kbps to ~50kbps, and the chunks ranged from 300+ to 0. Ill get smooth playback on occasion for 20-40 seconds then choppy. I do some see errors in the console. I setup a hotspot with my phone, and have the same issue - but its also not that great of a connection ~20mbps

Let me know if you have any suggestions on possible problems, or if there is some "poor internet tweaks". 

[137869][I][helixfuncs.h:124] playChunk(): [helixfuncs] Sync found at 0x003F
[137876][I][helixfuncs.h:170] playChunk(): [helixfuncs] MP3Decode error -9
[137882][I][helixfuncs.h:63] helixInit(): [helixfuncs] helixInit called for audi                                                                                                o/mpeg
[138146][I][helixfuncs.h:124] playChunk(): [helixfuncs] Sync found at 0x0047
[138159][I][helixfuncs.h:178] playChunk(): [helixfuncs] Bitrate     is 128000
[138160][I][helixfuncs.h:179] playChunk(): [helixfuncs] Samprate    is 44100
[138162][I][helixfuncs.h:180] playChunk(): [helixfuncs] Channels    is 2
[138169][I][helixfuncs.h:181] playChunk(): [helixfuncs] Bitpersamp  is 16
[138175][I][helixfuncs.h:182] playChunk(): [helixfuncs] Outputsamps is 2304
[138759][I][main.cpp:1277] showstreamtitle(): [main] Streamtitle found, 61 bytes
[138759][I][main.cpp:1278] showstreamtitle(): [main] StreamTitle='Elton John - S                                                                                                tep Into Christmas';StreamNext='';

aly-fly, 2023-12-01:
https://github.com/Edzelf/ESP32Radio-V2/issues/75#issuecomment-1755178480

1. Use local station.
2. Increase buffer size.

Edzelf, 2023-12-05:
The decode error is normal at the start of a new station (when using software mp3 decoder).  I have no cure for the drop in the network input stream.