Edzelf/ESP32Radio-V2 #133: Guru Meditation Error when starting

arminth, 2023-12-25:
Hi,
Merry Christmas!

I just wanted to upgrade my rigg to the new version and I am running into severe issues.
I have succesfully uploaded both the firware and the filesystem. When starting, system runs into a Guru Meditation error Core 1 paniced.
I have tried severa ESP32. I also did a flash erase before reflashing. Also reduced the Log-Level from 5 to 0. Nothing helped.

Here is the output (I took this snapshot with an ESP not hooked up to the display. Same with a different one that is hooked up.)

`Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3366][E][main.cpp:2689] setup(): [main] Display not activated
[ 15958][E][main.cpp:749] nvsgetstr(): [main] nvs_get_str failed 1102 for key preset, keylen is 6, len is 150!
[ 15959][E][main.cpp:750] nvsgetstr(): [main] Contents: 
[ 15964][E][AsyncTCP.cpp:729] connect(): error: -16
[ 15967][E][main.cpp:1472] connecttohost(): [main] Request  failed!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400ea983  PS      : 0x00060d30  A0      : 0x800d73d0  A1      : 0x3ffcef70
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x8d3d4800  A5      : 0x3ffcef60  
A6      : 0x3ffc60d4  A7      : 0x00000000  A8      : 0x800d55ec  A9      : 0x3ffcece0
A10     : 0x3ffced18  A11     : 0x00000004  A12     : 0x3f400124  A13     : 0x000005c0  
A14     : 0x3f400e40  A15     : 0x3f400ea8  SAR     : 0x00000019  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000009  LBEG    : 0x4008bb95  LEND    : 0x4008bba5  LCOUNT  : 0xffffffff  

Backtrace: 0x400ea980:0x3ffcef70 0x400d73cd:0x3ffcef90 0x400d87c4:0x3ffcefb0 0x400eeae9:0x3ffcefd0
`
Hope you have an idea!

Cheers
Armin

arminth, 2023-12-25:
Here a full log-dump on level 5:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (516) esp_core_dump_flash: No core dump f�ѥѥ���found!
E (516) esp_core_dump_flash: No core dump partition found!
[     8][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3042][I][main.cpp:2582] setup(): [main] Version Thu, 14 Dec 2023 09:00:00 GMT.  Free memory 110580
[  3044][I][main.cpp:2583] setup(): [main] Display type is BLUETFT
[  3140][I][main.cpp:2593] setup(): [main] SPIFFS is okay, space 1016801, used 33383
[  3143][I][main.cpp:2614] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3146][I][main.cpp:2614] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3155][I][main.cpp:2614] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1114112
[  3168][I][main.cpp:2486] fillkeylist(): [main] Read 0 keys from NVS
[  3172][I][main.cpp:1910] readIOprefs(): [main] pin_spi_sck set to 18
[  3177][I][main.cpp:1910] readIOprefs(): [main] pin_spi_miso set to 19
[  3183][I][main.cpp:1910] readIOprefs(): [main] pin_spi_mosi set to 23
[  3200][I][main.cpp:2655] setup(): [main] GPIO0 is HIGH
[  3210][I][main.cpp:2655] setup(): [main] GPIO2 is HIGH
[  3220][I][main.cpp:2655] setup(): [main] GPIO4 is HIGH
[  3230][I][main.cpp:2655] setup(): [main] GPIO5 is HIGH
[  3240][I][main.cpp:2655] setup(): [main] GPIO12 is HIGH
[  3250][I][main.cpp:2655] setup(): [main] GPIO13 is HIGH
[  3260][I][main.cpp:2655] setup(): [main] GPIO14 is HIGH
[  3270][I][main.cpp:2655] setup(): [main] GPIO15 is HIGH
[  3280][I][main.cpp:2655] setup(): [main] GPIO16 is HIGH
[  3290][I][main.cpp:2655] setup(): [main] GPIO17 is HIGH
[  3300][I][main.cpp:2655] setup(): [main] GPIO18 is HIGH
[  3310][I][main.cpp:2655] setup(): [main] GPIO19 is HIGH
[  3320][I][main.cpp:2655] setup(): [main] GPIO21 is HIGH
[  3330][I][main.cpp:2655] setup(): [main] GPIO22 is HIGH
[  3340][I][main.cpp:2655] setup(): [main] GPIO23 is HIGH
[  3350][I][main.cpp:2655] setup(): [main] GPIO25 is HIGH
[  3360][I][main.cpp:2655] setup(): [main] GPIO26 is HIGH
[  3370][I][main.cpp:2655] setup(): [main] GPIO27 is HIGH
[  3380][I][main.cpp:2655] setup(): [main] GPIO32 is HIGH
[  3390][I][main.cpp:2655] setup(): [main] GPIO33 is HIGH
[  3400][I][main.cpp:2655] setup(): [main] GPIO34 is LOW, probably no PULL-UP
[  3410][I][main.cpp:2655] setup(): [main] GPIO35 is LOW, probably no PULL-UP
[  3420][I][main.cpp:2655] setup(): [main] GPIO39 is LOW, probably no PULL-UP
[  3423][I][main.cpp:2672] setup(): [main] Start BLUETFT display
[  3423][E][main.cpp:2689] setup(): [main] Display not activated
[  3427][I][main.cpp:2278] mk_lsan(): [main] Create list with acceptable WiFi networks
[  3435][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Nase
[  3441][I][main.cpp:2308] mk_lsan(): [main] Added Nase to list of networks
[  3975][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  4076][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[  4076][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
[  4575][I][main.cpp:865] listNetworks(): [main] Scan Networks
[ 11189][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 10
[ 11189][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 11194][I][main.cpp:867] listNetworks(): [main] Scan completed
[ 11199][I][main.cpp:875] listNetworks(): [main] Number of available networks: 10
[ 11207][I][main.cpp:893] listNetworks(): [main]  1 - Garage                    Signal: -68 dBm, Encryption WPA2_PSK, Not accepttable
[ 11218][I][main.cpp:893] listNetworks(): [main]  2 - Nase                      Signal: -71 dBm, Encryption WPA_WPA2_PSK, Acceptable
[ 11230][I][main.cpp:893] listNetworks(): [main]  3 - Nase_CNC                  Signal: -74 dBm, Encryption WPA2_PSK, Acceptable
[ 11241][I][main.cpp:893] listNetworks(): [main]  4 - kueche                    Signal: -75 dBm, Encryption WPA_WPA2_PSK, Not accepttable
[ 11253][I][main.cpp:893] listNetworks(): [main]  5 - Nase3                     Signal: -76 dBm, Encryption WPA2_PSK, Acceptable
[ 11265][I][main.cpp:893] listNetworks(): [main]  6 - Stupsi                    Signal: -81 dBm, Encryption WPA2_PSK, Not accepttable
[... 32 lines cut ...]
[ 16526][E][main.cpp:1472] connecttohost(): [main] Request  failed!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400edf7b  PS      : 0x00060d30  A0      : 0x800d88ec  A1      : 0x3ffcef70
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x4f393946  A5      : 0x3ffcef50  
A6      : 0x3ffcef50  A7      : 0x00000080  A8      : 0x800d6460  A9      : 0x3ffc60d4
A10     : 0x3ffced08  A11     : 0x00000004  A12     : 0x3f400124  A13     : 0x000005c0  
A14     : 0x3f401e80  A15     : 0x3f401f1f  SAR     : 0x00000019  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000009  LBEG    : 0x4008bb95  LEND    : 0x4008bba5  LCOUNT  : 0xffffffff  

Backtrace: 0x400edf78:0x3ffcef70 0x400d88e9:0x3ffcef90 0x400da1cc:0x3ffcefb0 0x400f2369:0x3ffcefd0

[... cut ...] VS1053.cpp:216
  #1  0x400d88e9:0x3ffcef90 in spfuncs() at src/main.cpp:3352
      (inlined by) spfuncs() at src/main.cpp:3327
  #2  0x400da1cc:0x3ffcefb0 in loop() at src/main.cpp:3468
  #3  0x400f2369:0x3ffcefd0 in loopTask(void*) at C:/Users/armin/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: 38466b731bd5ce4d

E (16765) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jun  8 2016 00:22:57
```

trip5, 2023-12-25:
Just want to add here.  I'm encountering the exact same error with boards and parts that worked on the old version...

I'm re-using the older version just so I can still do something... (Having issues not needed to be discussed here.)

trip5, 2023-12-25:
I did clear up the dump error by kind of following this:

https://community.platformio.org/t/reset-after-platform-espressif32-upgrading-to-6-2-0/33946

...had to reduce the spiffs partition by 10000 and add a dump partition... but that didn't seem to help my particular situation.  @arminth Try it out and see if it helps.

radiopart.csv file:
````
# Special 4MB partition map.  app0/1 have some more space,
# SPIFFS has less space than default setting
#
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x170000,
app1,     app,  ota_1,   0x180000,0x170000,
spiffs,   data, spiffs,  0x2F0000,0x100000,
coredump, data, coredump,0x3F0000,0x10000,
#                END =   0x400000
````

arminth, 2023-12-26:
Thanks for the heads up on the coredump partition. That solved the core dump fault.
But still running into the Guru Meditation Error Core 1 panic'ed. :(

chudsp87, 2023-12-26:
i'm just getting started looking at this code so i'm not familiar with it yet, but ```EXCVADDR: 0x00000009``` suggests a dereferenced null pointer. Stack trace points to the `VS1053::setVolume` function; anything changed there? 

Could also be due to trying to write to protected area in memory or other failure in the write (bad connection; bad hardware). No sure if this is anymore helpful than what you might've already found on google, but hopefully it helps

Edzelf, 2023-12-26:
The error was caused by the VS1053 module.  If the list with preferences was empty, the object was not created, so setvol crashed.  Replece main.cpp and vs1053.c with the new versions.

trip5, 2023-12-26:
It seems to work now!

arminth, 2023-12-29:
I can definetly report, it's working again!
Thanks a lot Ed and happy new year!

closed 2023-12-29 by arminth