Edzelf/ESP32Radio-V2 #134: LCD_1602 How to display debug info

kpiwo, 2024-01-01:
Hi, what changes in the code should I make to get on LCD_1602 debug information such as Connecting to Wifi status or ip address.

aly-fly, 2024-01-01:
Best way forward: just abandon the idea if 2x16 char LCD, it is useless anyway. 
Use standard graphic LCD that is recommended and your experience will be waaay better. :)
p.s. HNY 2024 :)

kpiwo, 2024-01-01:
Thank you. Happy New Year. 
I'm using some bigger LCD 2x24 char, because it fits into classic Diora stereo tuner. To make display to work I use LCD1602 library with some modifications. This Diora stereo tuner: 
[image]
That's why I need to see what IP address I get when esp32 radio is connecting to network.

aly-fly, 2024-01-03:
Search in the code where IP address is printed onto the graphic LCD during boot.
Take code where print-out to your LCD type happen and copy to the initialization part. Add text with IP data.

FlintEastwood, 2024-01-13:
Please take a look in my fork:
https://github.com/FlintEastwood/ESP32Radio-V2

I already have implemented this with tftlog messages and some more stuff too for the LCD1602.
My IPdio is successfully revived now. :)