Edzelf/ESP32Radio-V2 #135: Accelerated playback

SP5MI, 2024-01-15:
Hi. When I'm trying to play streams from open.fm server, it sounds like 2x or 3x faster (stream.open.fm/378, stream.open.fm/368 for examle). Streams from other servers are ok. I'm using PCM5102 I2S module. Here is my serial log: 

D: 49140/93244 - Single click
D: 47092/93168 - nextPreset is 8
D: 47092/93168 - Radiofuncs cmd is 1
D: 45044/95304 - MP3Decode error -6
D: 45044/95304 - helixInit called for audio/mpeg, e is 255, d is 255
D: 45044/97720 - Sync found at 0x0055
D: 45044/97924 - MP3Decode error -2
D: 45044/97924 - helixInit called for audio/mpeg, e is 255, d is 255
D: 45044/97808 - Stopping client
D: 45044/100564 - Sync found at 0x0080
D: 45044/100564 - Playtask stop song
D: 59380/103000 - MP3 host error Connection aborted
D: 59380/103000 - Host disconnected
D: 59380/103068 - Connect to host stream.open.fm/378
D: 57332/100176 - Connected to host at 193.222.135.71 on port 80
D: 59380/102812 - send GET command
D: 57332/100944 - Request stream.open.fm/378 failed!
D: 53236/97968 - Switch to HEADER
D: 51188/95488 - Headerline: Content-Type: audio/aacp
D: 51188/95456 - Headerline: icy-metadata:1
D: 51188/95252 - Headerline: icy-name:378 (icy-br:48)
D: 47092/93144 - Headerline: icy-pub:0
D: 47092/93144 - Headerline: Server: Icecast 2.4.0-kh4
D: 47092/93144 - Headerline: Cache-Control: no-cache, no-store
D: 47092/93144 - Headerline: Access-Control-Allow-Origin: *
D: 47092/93144 - Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: 47092/93144 - Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: 47092/93144 - Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: 47092/93144 - Headerline: icy-metaint:16000
D: 47092/92984 - Switch to DATA, bitrate is 0, metaint is 16000
D: 47092/97976 - Playtask start song
D: 59380/102860 - helixInit called for audio/aacp, e is 255, d is 255
D: 59380/102860 - Sync found at 0x0000
D: 59380/102860 - Bitrate     is 36348
D: 59380/102860 - Samprate    is 22050
D: 59380/102860 - Channels    is 1
D: 59380/102860 - Bitpersamp  is 16
D: 49140/97988 - Outputsamps is 1024
D: 47092/93084 - Streamtitle found, 15 bytes
D: 47092/93084 - StreamTitle='';

Edzelf, 2024-01-15:
This is a station with mono signal (D: 59380/102860 - Channels is 1).  I know that there is a bug in the software concerning this matter, but I can't nail it down.

SP5MI, 2024-01-16:
Thank you Ed for a quick response. I found some workaround - open.fm is just a re-streaming service. In the sources of his website  i found links to original streams sending by radio stations, witch are in stereo mode. And now everything works perfectly. If I could suggest something, maybe it is a good idea to add some note to documentation about this bug. If someone wants to play with mono stations, then they will consider buying VS1053 rather than i2s module. Anyway, thank you for a great project. Now I can listening my favorite radio stations, even when I'm outside the radio waves range.

closed 2024-02-08 by SP5MI

Edzelf, 2024-02-20:
Mono should work in the latest version.