Edzelf/ESP32Radio-V2 #140: Display shows ESP32-Radio

Szvzs, 2024-02-07:
I have an issue, my project connects to wifi, i can access the web platform but my display says ESP32-Radio, i cannot hear anything on speaker. It does not show any info on the station playing and i can access only one station from the web server. This is from arduino ide serial monitor but i uploaded with VSC.

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3042][I][main.cpp:2584] setup(): [main] Version Thu, 26 Dec 2023 14:30:00 GMT.  Free memory 110580
[  3044][I][main.cpp:2585] setup(): [main] Display type is BLUETFT
[  3138][I][main.cpp:2595] setup(): [main] SPIFFS is okay, space 1016801, used 33383
[  3142][I][main.cpp:2616] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3144][I][main.cpp:2616] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3154][I][main.cpp:2616] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1114112
[  3166][I][main.cpp:2488] fillkeylist(): [main] Read 45 keys from NVS
[  3177][I][main.cpp:1911] readIOprefs(): [main] pin_enc_clk set to 25
[  3178][I][main.cpp:1911] readIOprefs(): [main] pin_enc_dt set to 26
[  3183][I][main.cpp:1911] readIOprefs(): [main] pin_enc_sw set to 27
[  3189][I][main.cpp:1911] readIOprefs(): [main] pin_tft_cs set to 15
[  3195][I][main.cpp:1911] readIOprefs(): [main] pin_tft_dc set to 2
[  3201][I][main.cpp:1911] readIOprefs(): [main] pin_vs_cs set to 5
[  3207][I][main.cpp:1911] readIOprefs(): [main] pin_vs_dcs set to 32
[  3213][I][main.cpp:1911] readIOprefs(): [main] pin_vs_dreq set to 35
[  3219][I][main.cpp:1911] readIOprefs(): [main] pin_spi_sck set to 18
[  3225][I][main.cpp:1911] readIOprefs(): [main] pin_spi_miso set to 19
[  3231][I][main.cpp:1911] readIOprefs(): [main] pin_spi_mosi set to 23
[  3248][I][main.cpp:2657] setup(): [main] GPIO0 is HIGH
[  3258][I][main.cpp:2657] setup(): [main] GPIO2 is HIGH
[  3268][I][main.cpp:2657] setup(): [main] GPIO4 is HIGH
[  3278][I][main.cpp:2657] setup(): [main] GPIO5 is HIGH
[  3288][I][main.cpp:2657] setup(): [main] GPIO12 is HIGH
[  3298][I][main.cpp:2657] setup(): [main] GPIO13 is HIGH
[  3308][I][main.cpp:2657] setup(): [main] GPIO14 is HIGH
[  3318][I][main.cpp:2657] setup(): [main] GPIO15 is HIGH
[  3328][I][main.cpp:2657] setup(): [main] GPIO16 is HIGH
[  3338][I][main.cpp:2657] setup(): [main] GPIO17 is HIGH
[  3348][I][main.cpp:2657] setup(): [main] GPIO18 is HIGH
[  3358][I][main.cpp:2657] setup(): [main] GPIO19 is HIGH
[  3368][I][main.cpp:2657] setup(): [main] GPIO21 is HIGH
[  3378][I][main.cpp:2657] setup(): [main] GPIO22 is HIGH
[  3388][I][main.cpp:2657] setup(): [main] GPIO23 is HIGH
[  3398][I][main.cpp:2657] setup(): [main] GPIO25 is HIGH
[  3408][I][main.cpp:2657] setup(): [main] GPIO26 is HIGH
[  3418][I][main.cpp:2657] setup(): [main] GPIO27 is HIGH
[  3428][I][main.cpp:2657] setup(): [main] GPIO32 is HIGH
[  3438][I][main.cpp:2657] setup(): [main] GPIO33 is HIGH
[  3448][I][main.cpp:2657] setup(): [main] GPIO34 is LOW, probably no PULL-UP
[  3458][I][main.cpp:2657] setup(): [main] GPIO35 is LOW, probably no PULL-UP
[  3468][I][main.cpp:2657] setup(): [main] GPIO39 is LOW, probably no PULL-UP
[  3469][I][main.cpp:1757] readprogbuttons(): [main] gpio_00 will execute uppreset = 1
[  3472][I][main.cpp:1757] readprogbuttons(): [main] gpio_12 will execute upvolume = 2
[  3480][I][main.cpp:1757] readprogbuttons(): [main] gpio_13 will execute downvolume = 2
[  3488][I][main.cpp:1757] readprogbuttons(): [main] gpio_14 will execute stop
[  3495][I][main.cpp:1757] readprogbuttons(): [main] gpio_17 will execute resume
[  3502][I][main.cpp:1757] readprogbuttons(): [main] gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
[  3513][I][main.cpp:2674] setup(): [main] Start BLUETFT display
[  4302][I][main.cpp:2280] mk_lsan(): [main] Create list with acceptable WiFi networks
[  4302][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: TP-Link_98F3
[  4305][I][main.cpp:2310] mk_lsan(): [main] Added TP-Link_98F3 to list of networks
[  4313][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: TP-Link_98F3
[  4320][I][main.cpp:2310] mk_lsan(): [main] Added TP-Link_98F3 to list of networks
[  4328][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: SSID2
[  4333][I][main.cpp:2310] mk_lsan(): [main] Added SSID2 to list of networks
[  4857][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  4953][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[  4954][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
[  5453][I][main.cpp:866] listNetworks(): [main] Scan Networks
[ 11466][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 3
[ 11467][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 11471][I][main.cpp:868] listNetworks(): [main] Scan completed
[ 11477][I][main.cpp:876] listNetworks(): [main] Number of available networks: 3
[ 11484][I][main.cpp:894] listNetworks(): [main]  1 - disabled                  Signal: -70 dBm, Encryption WPA_WPA2_PSK, Not accepttable
[ 11496][I][main.cpp:894] listNetworks(): [main]  2 - TP-Link_98F3              Signal: -75 dBm, Encryption WPA_WPA2_PSK, Acceptable
[ 11508][I][main.cpp:894] listNetworks(): [main]  3 - Anna                      Signal: -92 dBm, Encryption WPA2_PSK, Not accepttable
[ 11519][I][main.cpp:898] listNetworks(): [main] End of list
[... 32 lines cut ...]
[ 24234][I][main.cpp:4388] gettime(): [main] TOD synced
[ 26235][I][main.cpp:2836] setup(): [main] Playtask started
[ 26452][I][main.cpp:967] nextPreset(): [main] nextPreset is 6
[ 26453][I][main.cpp:3406] radiofuncs(): [main] Radiofuncs cmd is 1
[ 26454][I][main.cpp:1401] connecttohost(): [main] Connect to host icecast.omroep.nl:80/radio1-bb-mp3
[ 26721][I][main.cpp:2543] onConnect(): [main] Connected to host at 145.58.53.167 on port 80
[ 26765][I][main.cpp:1462] connecttohost(): [main] send GET command
[ 26825][I][main.cpp:3654] handlebyte_ch(): [main] Switch to HEADER
[ 26825][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: Server: Icecast 2.4.0-kh22
[ 26828][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: Connection: Close
[ 26835][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: Pragma: no-cache
[ 26842][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: Expires: Thu, 19 Nov 1981 08:52:00 GMT
[ 26851][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: Cache-Control: no-store, no-cache, private
[ 26860][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: Content-Type: audio/mpeg
[ 26868][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: icy-br: 192
[ 26874][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: ice-audio-info: samplerate=48000;channels=2;bitrate=192
[ 26885][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: icy-genre: Talk
[ 26892][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: icy-metadata: 1
[ 26899][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: icy-name: NPO Radio1
[ 26906][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: icy-pub: 0
[ 26913][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: icy-url: http://www.radio1.nl
[ 26921][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: icy-metaint: 16000
[ 26928][I][main.cpp:3745] handlebyte_ch(): [main] Switch to DATA, bitrate is 192, metaint is 16000
[ 27053][I][main.cpp:1295] showstreamtitle(): [main] Streamtitle found, 52 bytes
[ 27053][I][main.cpp:1296] showstreamtitle(): [main] StreamTitle='NPO Radio 1 - Nieuws en Co - NOS, NTR';

Edzelf, 2024-02-07:
Are you sure you are using the VS1053?  Check your config.h.
I miss these lines in the logging:
```
[ 28079][I][VS1053.cpp:129] testComm(): [VS1053] Slow SPI, Testing VS1053 read/write registers...
[ 28338][I][VS1053.cpp:129] testComm(): [VS1053] Fast SPI, Testing VS1053 read/write registers again...

```

Szvzs, 2024-02-08:
I checked the config and VS1053 is selected and Bluetft. I managed to save the output from vsc serial monitor (yesterday i noticed baud rate was not ok so i had a lot of bad characters so this is from vsc). At the moment i do not have the VS1053 wired , only the display. 

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (517) esp_core_dump_flash: No core dump
�ѥѥ�� found!
E (517) esp_core_dump_flash: No core dump partition found!
[     8][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3042][I][main.cpp:2584] setup(): [main] Version Thu, 26 Dec 2023 14:30:00 GMT.  Free memory 110580
[  3044][I][main.cpp:2585] setup(): [main] Display type is BLUETFT
[  3138][I][main.cpp:2595] setup(): [main] SPIFFS is okay, space 1016801, used 33383
[  3142][I][main.cpp:2616] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3144][I][main.cpp:2616] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3154][I][main.cpp:2616] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1114112
[  3166][I][main.cpp:2488] fillkeylist(): [main] Read 45 keys from NVS
[  3177][I][main.cpp:1911] readIOprefs(): [main] pin_enc_clk set to 25
[  3178][I][main.cpp:1911] readIOprefs(): [main] pin_enc_dt set to 26
[  3183][I][main.cpp:1911] readIOprefs(): [main] pin_enc_sw set to 27
[  3189][I][main.cpp:1911] readIOprefs(): [main] pin_tft_cs set to 15
[  3195][I][main.cpp:1911] readIOprefs(): [main] pin_tft_dc set to 2
[  3201][I][main.cpp:1911] readIOprefs(): [main] pin_vs_cs set to 5
[  3207][I][main.cpp:1911] readIOprefs(): [main] pin_vs_dcs set to 32
[  3213][I][main.cpp:1911] readIOprefs(): [main] pin_vs_dreq set to 35
[  3219][I][main.cpp:1911] readIOprefs(): [main] pin_spi_sck set to 18
[  3225][I][main.cpp:1911] readIOprefs(): [main] pin_spi_miso set to 19
[  3231][I][main.cpp:1911] readIOprefs(): [main] pin_spi_mosi set to 23
[  3248][I][main.cpp:2657] setup(): [main] GPIO0 is HIGH
[  3258][I][main.cpp:2657] setup(): [main] GPIO2 is HIGH
[  3268][I][main.cpp:2657] setup(): [main] GPIO4 is HIGH
[  3278][I][main.cpp:2657] setup(): [main] GPIO5 is HIGH
[  3288][I][main.cpp:2657] setup(): [main] GPIO12 is HIGH
[  3298][I][main.cpp:2657] setup(): [main] GPIO13 is HIGH
[  3308][I][main.cpp:2657] setup(): [main] GPIO14 is HIGH
[  3318][I][main.cpp:2657] setup(): [main] GPIO15 is HIGH
[  3328][I][main.cpp:2657] setup(): [main] GPIO16 is HIGH
[  3338][I][main.cpp:2657] setup(): [main] GPIO17 is HIGH
[  3348][I][main.cpp:2657] setup(): [main] GPIO18 is HIGH
[  3358][I][main.cpp:2657] setup(): [main] GPIO19 is HIGH
[  3368][I][main.cpp:2657] setup(): [main] GPIO21 is HIGH
[  3378][I][main.cpp:2657] setup(): [main] GPIO22 is HIGH
[  3388][I][main.cpp:2657] setup(): [main] GPIO23 is HIGH
[  3398][I][main.cpp:2657] setup(): [main] GPIO25 is HIGH
[  3408][I][main.cpp:2657] setup(): [main] GPIO26 is HIGH
[  3418][I][main.cpp:2657] setup(): [main] GPIO27 is HIGH
[  3428][I][main.cpp:2657] setup(): [main] GPIO32 is HIGH
[  3438][I][main.cpp:2657] setup(): [main] GPIO33 is HIGH
[  3448][I][main.cpp:2657] setup(): [main] GPIO34 is LOW, probably no PULL-UP
[  3458][I][main.cpp:2657] setup(): [main] GPIO35 is LOW, probably no PULL-UP
[  3468][I][main.cpp:2657] setup(): [main] GPIO39 is LOW, probably no PULL-UP
[  3469][I][main.cpp:1757] readprogbuttons(): [main] gpio_00 will execute uppreset = 1
[  3472][I][main.cpp:1757] readprogbuttons(): [main] gpio_12 will execute upvolume = 2
[  3480][I][main.cpp:1757] readprogbuttons(): [main] gpio_13 will execute downvolume = 2
[  3488][I][main.cpp:1757] readprogbuttons(): [main] gpio_14 will execute stop
[  3495][I][main.cpp:1757] readprogbuttons(): [main] gpio_17 will execute resume
[  3502][I][main.cpp:1757] readprogbuttons(): [main] gpio_34 will execute station = icecast.omroep.nl:80/radio1-bb-mp3
[  3513][I][main.cpp:2674] setup(): [main] Start BLUETFT display
[  4302][I][main.cpp:2280] mk_lsan(): [main] Create list with acceptable WiFi networks
[  4302][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: TP-Link_98F3
[  4305][I][main.cpp:2310] mk_lsan(): [main] Added TP-Link_98F3 to list of networks
[  4313][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: TP-Link_98F3
[  4320][I][main.cpp:2310] mk_lsan(): [main] Added TP-Link_98F3 to list of networks
[  4328][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: SSID2
[... 44 lines cut ...]
[ 24534][I][main.cpp:4388] gettime(): [main] TOD synced
����������������������������������������������������������������(): [main] Playtask started
[ 26752][I][main.cpp:967] nextPreset(): [main] nextPreset is 6
[ 26753][I][main.cpp:3406] radiofuncs(): [main] Radiofuncs cmd is 1
[ 26754][I][main.cpp:1401] connecttohost(): [main] Connect to host icecast.omroep.nl:80/radio1-bb-mp3
[ 26979][I][main.cpp:2543] onConnect(): [main] Connected to host at 145.58.52.167 on port 80
[ 27064][I][main.cpp:1462] connecttohost(): [main] send GET command
[ 27125][I][main.cpp:3654] handlebyte_ch(): [main] Switch to HEADER
[ 27125][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: Server: Icecast 2.4.0-kh22
[... cut ...]
[ 27206][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: icy-pub: 0
[ 27212][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: icy-url: http://www.radio1.nl
[ 27221][I][main.cpp:3675] handlebyte_ch(): [main] Headerline: icy-metaint: 16000
[ 27228][I][main.cpp:3745] handlebyte_ch(): [main] Switch to DATA, bitrate is 192, metaint is 16000
[ 27401][I][main.cpp:1295] showstreamtitle(): [main] Streamtitle found, 55 bytes
[ 27401][I][main.cpp:1296] showstreamtitle(): [main] StreamTitle='NPO Radio 1 - NOS Radio 1 Journaal - NOS';

Edzelf, 2024-02-08:
I made a typo in platformio.ini.  The baudrate is set to 115600.  That should be 115200.
I still dont understand why (in your set-up) the debuglines with "`[VS1053]`" are missing.
I am pretty sure that if you can find out why, the problem will be solved.
Add some extra ESP_LOGI ( VTAG, "xxxxx" ) lines in VS1053.begin() to find out what is wrong.

Szvzs, 2024-02-08:
For clarification: 

1. i written the program with vsc without any phisical connections made
2. made connection only for vs1053 but did not select dummytft so i think it was looking for the other connections
3. connected only tft so it was looking for the other connections for vs1053 i guess(the first serial output)
4. i did all the connections and it works (i will not paste the serial output but i found logging for the vs1053)
5. Thank you and sorry for the "newbie" questions!

As an off topic question, do you have second encoder implemented for station preset change? I have an old radio sokol 304 that i am trying to transform and it has two potentiometer type controls in front which i would like to use for full access of radio functions.

closed 2024-02-08 by Szvzs