Edzelf/ESP32Radio-V2 #142: Problems Configuering SPDIF Output

MrTwisterFister69, 2024-03-10:
Hey,

I really like this project! Great work! I'd love to build it myself, but I'm having trouble getting the SPDIF output to work. Here's my config from the web interface:

    Daylight saving time offset: 1 hour
    UTC offset: 1 hour
    Time server: pool.ntp.org
    GPIO settings:
        gpio_00: upper preset (set to 1)
        gpio_14: stop
        gpio_17: resume
        gpio_34: station (set to icecast.omroep.nl:80/radio1-bb-mp3)
    Rotary encoder pins:
        CLK: 25
        DT: 26
        SW: 27
    SPDIF pin: 2
    Presets:
        Preset 00: http://icecast.ndr.de/ndr/njoy/live/mp3/128/stream.mp3 (N-Joy)
    Tone settings: all set to 0
    Volume: 100
    Wi-Fi settings:
        Wi-Fi 00: /***
        Wi-Fi 01: SSID2/PASSWD2

Also, I'm wondering if it's possible to ditch the rotary encoder and just rely on buttons for control.

Thanks in advance, and sorry for my improvised English.

Edzelf, 2024-03-11:
Are you sure you have defined "pin_i2s_spdif" in the config page of the web interface?
If you remove the definitions of "pin_enc_clk", "pin_enc_clk" and "pin_enc_clk" from the config page in the web interface, the rotary encoder is not activated.
The LED in the SPDIF transmitter should light up.  You can make a simple "blink" program using the pin connected to the SPDIF transmitter to test it.
Alway show the serial debug output, so we can help to find the problem.

MrTwisterFister69, 2024-03-17:
Is there a tutorial on how to use the Spdif support? My web interface configs look like this right now and i dont get any sound:
($)pin_i2s_bck = 5                                      # GPIO Pin number for I2S "BCK"
($)pin_i2s_din = 34                                     # GPIO Pin number for I2S "DIN"
($)pin_i2s_lck = 32                                     # GPIO Pin number for I2S "L(R)CK"
#
clk_dst = 1                                          # Offset during daylight saving time (hours)
clk_offset = 1                                       # Offset with respect to UTC in hours
clk_server = pool.ntp.org                            # Time server to be used
#
gpio_00 = uppreset = 1
gpio_12 = upvolume = 2
gpio_13 = downvolume = 2
gpio_14 = stop
gpio_17 = resume
gpio_34 = station = icecast.omroep.nl:80/radio1-bb-mp3

mqttbroker = none
mqttpasswd = none
mqttport = 1883
mqttprefix = none
mqttuser = none
#

pin_i2s_spdif = 5
pin_ir = 35                                          # GPIO Pin number for IR receiver VS1838B
pin_tft_cs = 15                                      # GPIO Pin number for TFT "CS"
pin_tft_dc = 2                                       # GPIO Pin number for TFT "DC"

#
preset = 1
preset_00 = 109.206.96.34:8100                       #  0 - NAXI LOVE RADIO, Belgrade, Serbia
preset_01 = airspectrum.cdnstream1.com:8114/1648_128 #  1 - Easy Hits Florida 128k
preset_02 = stream.hitfm.de:9050/live                #  2 - 89 HIT FM - Munichs Power Station, 128k
preset_03 = airspectrum.cdnstream1.com:8000/1261_192 #  3 - Magic Oldies Florida
preset_04 = airspectrum.cdnstream1.com:8008/1604_128 #  4 - Magic 60s Florida 60s Classic Rock
preset_05 = us1.internet-radio.com:8105              #  5 - Classic Rock Florida - SHE Radio
preset_06 = icecast.omroep.nl:80/radio1-bb-mp3       #  6 - Radio 1, NL, 192k
preset_07 = 205.164.62.15:10032                      #  7 - 1.FM - GAIA, 64k
preset_08 = www.doowopradio.com:8000                 #  8 - Doo-Wop Radio, 96k
preset_09 = server-27.stream-server.nl:8192/stream   #  9 - 192 Radio Nederland, 320k
#
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
volume = 100
#
wifi_00 = ***/***
wifi_01 = SSID2/PASSWD2

Maybe I am just making a dumb mistake, but I am very confused at the moment.

Edzelf, 2024-03-18:
Read [this](https://github.com/Edzelf/ESP32-Radio/blob/master/doc/ESP32-radio%20buiding%20instructions.pdf).  See chapter 1.3.

yochiwarez, 2024-05-22:
Hi, I also haven't been able to get the SPDIF output to work, I've followed the guide to the letter. Please help

($)pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder CLK
($)pin_enc_dt = 26                                     # GPIO Pin number for rotary encoder DT
($)pin_enc_sw = 27                                     # GPIO Pin number for rotary encoder SW
($)pin_ir = 35                                     # GPIO Pin number for IR receiver VS1838B
($)pin_sd_cs = 21                                     # GPIO Pin number for SD card CS
($)pin_tft_dc = 2                                      # GPIO Pin number for TFT DC
($)pin_vs_cs = 5                                      # GPIO Pin number for VS1053 CS
($)pin_vs_dcs = 16                                     # GPIO Pin number for VS1053 DCS
($)pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 DREQ
#
clk_dst = 1
clk_offset = 1
clk_server = pool.ntp.org
#
gpio_00 = uppreset = 1
gpio_12 = upvolume = 2
gpio_13 = downvolume = 2
#
ir_40BF = upvolume = 2
ir_C03F = downvolume = 2
#
mqttbroker = none
mqttpasswd = none
mqttport = 1883
mqttprefix = none
mqttuser = none
#
pin_i2s_spdif = 15
#
preset = 0
preset_00 = relay.181.fm:8074                        #  0 - vibe of vegas
preset_01 = airspectrum.cdnstream1.com:8114/1648_128 #  1 - Easy Hits Florida 128k
preset_02 = us2.internet-radio.com:8050              #  2 - CLASSIC ROCK MIA WWW.SHERADIO.COM
preset_03 = airspectrum.cdnstream1.com:8000/1261_192 #  3 - Magic Oldies Florida
preset_04 = airspectrum.cdnstream1.com:8008/1604_128 #  4 - Magic 60s Florida 60s Classic Rock
preset_05 = us1.internet-radio.com:8105              #  5 - Classic Rock Florida - SHE Radio
preset_06 = icecast.omroep.nl:80/radio1-bb-mp3       #  6 - Radio 1, NL
preset_07 = 205.164.62.15:10032                      #  7 - 1.FM - GAIA, 64k
preset_08 = skonto.ls.lv:8002/mp3                    #  8 - Skonto 128k
preset_09 = 94.23.66.155:8106                        #  9 - *ILR CHILL and GROOVE
preset_10 = ihr/IHR_IEDM                             # 10 - iHeartRadio IHR_IEDM
preset_11 = ihr/IHR_TRAN                             # 11 - iHeartRadio IHR_TRAN
preset_12 = 109.206.96.34:8100                       # 12 - NAXI LOVE RADIO, Belgrade, Serbia
#
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
volume = 72
#
wifi_00 = ElPapuPro/******

------------------------

//***************************************************************************************************
// config.h                                                                                         *
//***************************************************************************************************
// Configuration definition for your particular set-up.                                             *
//***************************************************************************************************
//
#ifndef CONFIG_H
  //#define NAME "ESP32-Radio"                              // Define name of the radio, also AP SSID,
                                                            // also namespace in NVS.
                                                            // Default is "ESP32-Radio"

  //#define SDCARD                                          // Experimental: For SD card support (reading MP3-files)

[... 85 lines cut ...]
[ 15734][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Content-Type: audio/mpeg
[ 15742][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: icy-metaint: 16000
[ 15749][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: icy-genre: Easy Listening
[ 15757][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Instance-id: b4d017bb3d2a776a26ae2b10da8dc0cc
[ 15767][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Server: Streaming Server 9.0.7
[ 15776][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: icy-br: 128
[ 15783][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
[ 15792][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Pragma: no-cache
[ 15799][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Cache-Control: no-cache
[ 15807][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Connection: close
[ 15814][I][main.cpp:3625] handlebyte_ch(): [main] Switch to DATA, bitrate is 128 kbps, metaint is 16000
[ 15824][I][main.cpp:4474] playtask(): [main] Playtask start song
[ 15830][I][helixfuncs.h:114] helixInit(): [helixfuncs] helixInit called for audio/mpeg
[ 15915][I][helixfuncs.h:233] playChunk(): [helixfuncs] Sync found at 0x0000
[ 15938][I][helixfuncs.h:286] playChunk(): [helixfuncs] Bitrate     is 128000
[ 15945][I][helixfuncs.h:287] playChunk(): [helixfuncs] Samprate    is 44100
[ 15952][I][helixfuncs.h:288] playChunk(): [helixfuncs] Channels    is 2
[ 15959][I][helixfuncs.h:289] playChunk(): [helixfuncs] Bitpersamp  is 16
[ 15966][I][helixfuncs.h:290] playChunk(): [helixfuncs] Outputsamps is 2304
[ 16554][I][main.cpp:1224] showstreamtitle(): [main] Streamtitle found, 53 bytes
[ 16561][I][main.cpp:1225] showstreamtitle(): [main] StreamTitle='Dynoro & Gigi D'Agostino - In My Mind ';
[ 18389][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[ 18913][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[ 19451][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[ 20526][W][AsyncTCP.cpp:930] _poll(): pcb is NULL

Edzelf, 2024-05-23:
Try to make a "blink" program for GPIO 15 and check if the red LED in the led is blinking.

yochiwarez, 2024-05-23:
thank you for responding.
https://github.com/Edzelf/ESP32Radio-V2/assets/17991421/a24ab747-9fcf-4993-8e8c-ee110d74e232

fscii, 2024-06-19:
I also have no issue using a blink sketch on gpio 15.  I have same config.h as shown by other user above (except wifi credentials).  I also have no audio from SPDIF.  However, using this example from another library:
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-communication/a2dp/streams-a2dp-spdif

(simply changing proper pin to 15 for the ESP32-Radio hardware boards I got from PCBWay -- it does work.  So it isn't my hardware or config.  Something is definitely wrong with SPDIF code in ESP32-Radio V2

Edzelf, 2024-06-27:
I was able to reproduce the bug.  I am working on it....

dreamlayers, 2024-11-23:
This should probably be closed. The current version works with ESP32-WROOM-32U and a Neoteck NTK004a SPDIF DAC using optical input. Seems these fixes were probably in the Fri Jul 5 2024 commits. Setting of pin_config.bck_io_num and pin_config.ws_io_num from those commits was later removed, showing that I2S data output can function even when the clock outputs are not assigned to pins.