Edzelf/ESP32Radio-V2 #145: stop_mp3client() causes reboot

aly-fly, 2024-03-25:
All libraries are up to date (removed specific versions in the platformio.ini)
When switching the presets the function `stop_mp3client` is called which causes reboot of the system:
```
[ 19693][I][main.cpp:1225] showstreamtitle(): [main] StreamTitle='BILLY JOEL - MY LIFE';
[136929][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
Stack loopTask is 5808
Stack Playtask is  484
ADC reading is 0, filtered 0
0 IR interrupts seen
[147514][I][main.cpp:895] nextPreset(): [main] nextPreset is 7
[147516][I][main.cpp:3303] radiofuncs(): [main] Radiofuncs cmd is 1
[147523][I][main.cpp:1303] stop_mp3client(): [main] Stopping client

assert failed: tcp_update_rcv_ann_wnd IDF/components/lwip/lwip/src/core/tcp.c:951 (new_rcv_ann_wnd <= 0xffff)
Backtrace: ....
Rebooting...
```

After checking various variants I found that the `abort` command causes crash.
```
    //mp3client->close() ;                         // Causes memory leak! --> uncommenting this fixes the issue
    mp3client->abort() ;                           // This works better --> causes reboot
```
Using both lines fixes the issue and disconnection is smooth.

Edzelf, 2024-03-26:
Thanks, I will try that.

Edzelf, 2024-04-24:
Espressif platform version 6.6.0 is the latest up to now.  It should be okay to use it.
The SD directory in the project should be deleted, it was a left-over of a previous version of the radio.
The stack size needed for the SD card is a problem.  A bigger size may cause the webinterface to crash.