Edzelf/ESP32Radio-V2 #146: Audio stream stopps after changing preselected station , reboot need

thb584Git, 2024-04-16:
If have some issues with my ESp32-Radio design and need advice. Designed a PCB 2 years ago (you can find in this [post](https://github.com/Edzelf/ESP32Radio-V2/issues/61)). However my wife decided to stay with the installed kitchen radio so I put it on the shelf. Now there is demand to go "productive" an I installed the lasted version (not the release V1.0).

HW: NodeMCU-32S, pHAT_I²S-DAC(PCM5102), OLED Display SSD1306, rotary encoder + 4 Buttons

First issue, not really a problem just strange and maybe related. The define for FIXEDWIFI in config.h is ignored. The board always comes up with its predefined AP and IP. 

Modifying the preferences via web-interface works. After reboot It connects to my Wifi-AP and the audio stream is playing, but...

And this is the main issue. After turning on audio is o.k. Not changing the preselected station it will be playing for hours. On changing the station it keeps is playing as well but only if the station tuning was done a short time after boot. 

To be more recise, you can change the preselect several times until you stayed tuned to a station for some minutes. By switching to another station then the audio stops. Test feature throws e.g. "Free memory is 47092/98324, chunks in queue 0, bitrate 3426287 kbps". Behavior is the same using web commands, buttons or rotary encoder for preselect.

Volume Buttons and rotary encoders are still working and display information is as expected except no Stream content information. Audio will not work before a reset was applied.

Configuation:

```
//***************************************************************************************************
// config.h                                                                                         *
//***************************************************************************************************
//
//#define NAME "ESP32-Radio"                              // Define name of the radio, also AP SSID
                                                          // Default is "ESP32-Radio"

//#define SDCARD                                          // For SD card support (reading MP3-files)

#define FIXEDWIFI "DA*****/4*****!"   // Add a fixed SSID to the list

// Define (just one) type of MP3/AAC decoder
//#define DEC_VS1053                                        // Hardware decoder for MP3, AAC, OGG
//#define DEC_VS1003                                      // Hardware decoder for MP3
#define DEC_HELIX                                         // Software decoder for MP3, AAC. I2S output
//#define DEC_HELIX_INT                                   // Software decoder for MP3, AAC. DAC output
                                                          // Needs platform = espressif32@3.1.0 !!!!

// Define (just one) type of display.  See documentation.
//#define BLUETFT                                           // Works also for RED TFT 128x160
//#define ST7789                                          // 240x240 TFT
#define OLED1306                                        // 64x128 I2C OLED SSD1306
//#define OLED1309                                        // 64x128 I2C OLED SSD1309
//#define OLED1106                                        // 64x128 I2C OLED SH1106
//#define DUMMYTFT                                        // Dummy display
//#define LCD1602I2C                                      // LCD 1602 display with I2C backpack
//#define LCD2004I2C                                      // LCD 2004 display with I2C backpack
//#define ILI9341                                         // ILI9341 240*320
//#define NEXTION                                         // Nextion display. Uses UART 2 (pin 16 and 17)
//

// Define ZIPPYB5 if a ZIPPY B5 Side Switch is used instead of a rotary switch
///#define ZIPPYB5

// End of configuration parameters.

// DEC_HELIX_INT does not seem to work with higher versions of framework.  Do a test.
#ifdef DEC_HELIX_INT
  #if ESP_ARDUINO_VERSION_MAJOR >= 2
    #error Internal DAC will not work in this version of the platform.  Use espressif32@3.1.0.
  #endif
#endif

```

```
($)pin_i2s_bck = 5                                     # GPIO Pin number for I2S "BCK"
($)pin_i2s_din = 4                                     # GPIO Pin number for I2S "DIN"
($)pin_i2s_lck = 32                                    # GPIO Pin number for I2S "L(R)CK"
#
batt0 = 2600
batt100 = 2950
#
[... 142 lines cut ...]
[ 13228][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 202 - AUTH_FAIL
[ 13235][D][WiFiGeneric.cpp:1077] _eventCallback(): WiFi Reconnect Running
[ 13242][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 13776][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: DATALAN, BSSID: dc:39:6f:b0:62:f4, Channel: 1, Auth: WPA_WPA2_PSK
[ 13778][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 13824][V][WiFiGeneric.cpp:369] _arduino_event_cb(): STA Got New IP:192.168.0.126
[ 13825][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 13828][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.0.126, MASK: 255.255.255.0, GW: 192.168.0.1
[ 13838][I][WiFiMulti.cpp:177] run(): [WIFI] Connecting done.
[ 13843][D][WiFiMulti.cpp:178] run(): [WIFI] SSID: DATALAN
[ 13849][D][WiFiMulti.cpp:179] run(): [WIFI] IP: 192.168.0.126
[ 13854][D][WiFiMulti.cpp:180] run(): [WIFI] MAC: DC:39:6F:B0:62:F4
[ 13860][D][WiFiMulti.cpp:181] run(): [WIFI] Channel: 1
[ 18134][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 68396][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 68935][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 71184][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
Stack loopTask is 6004
Stack Playtask is  616
ADC reading is 0, filtered 20
16 IR interrupts seen
[ 78977][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 84017][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[ 89034][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
```

thb584Git, 2024-04-16:
Did more testing, built a second device using the latest release I had in my git local repo  (GITEA) from two years ago:

- WLAN AP in config.h was recocknized

- dubug gives much more information (was defined 0 that time)

  ```
  build_flags =
  	-DCORE_DEBUG_LEVEL=0
  	-Os
  ;	-Wl,-Map,ESP32-Radio.map
  ```

- but audio play issue exist here as well, same behavior

Here the Log-file for boot and one working preset change and a second one failing

```
D: Starting ESP32-radio running on CPU 1 at 240 MHz.
D: Version Sun, 15 May 2022 10:30:00 GMT.  Free memory 110580
D: Display type is OLED
D: SPIFFS is okay, space 1318001, used 33634
D: Found partition 'nvs     ' at offset 0x009000 with size    20480
D: Found partition 'otadata ' at offset 0x00E000 with size     8192
D: Found partition 'spiffs  ' at offset 0x290000 with size  1441792
D: Found partition 'coredump' at offset 0x3F0000 with size    65536
D: Read 46 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_scl set to 22
D: pin_tft_sda set to 21
D: pin_i2s_bck set to 5
D: pin_i2s_lck set to 32
D: pin_i2s_din set to 4
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is HIGH
D: GPIO35 is HIGH
D: GPIO39 is HIGH
D: gpio_00 will execute upvolume= 3
D: gpio_12 will execute mute
D: gpio_13 will execute uppreset= 1
D: gpio_34 will execute downpreset = 1
D: gpio_39 will execute downvolume= 3
D: Enable pin 35 for IR
D: Start OLED display
D: Init OLED 1306, I2C pins 21,22
[... 178 lines cut ...]
D: MP3Decode error -6
D: helixInit called for audio/mpeg, e is 255, d is 255
D: Sync found at 0x0120
D: Bitrate     is 128000
D: Samprate    is 48000
D: Channels    is 2
D: Bitpersamp  is 16
D: Outputsamps is 2304
D: Streamtitle found, 69 bytes
D: StreamTitle='SWR3 Nachrichten - auch zum Nachhören in der SWR3 App';
D: Publish to topic radio1/icy/streamtitle : SWR3 Nachrichten - auch zum Nachhören in der SWR3 App
D: updateNr 1 <= 8 to 7, relative is 0
D: updateNr result is 7
D: nextPreset is 7
D: Radiofuncs cmd is 1
D: Stopping client
D: Playtask stop song
D: Connect to host dispatcher.rndfnk.com/swr/swr1/rp/mp3/128/stream.mp3
D: Publish to topic radio1/preset : 7
D: Publish to topic radio1/playing : 0
D: Stack maintask  is 6200
D: Stack playtask  is 992
D: ADC reading is 765
D: 36 IR interrupts seen
```

Edzelf, 2024-04-16:
The FIXEDWIFI is just an (extra) entry in the list of WiFi networks.  Handy if the configured list is still empty.

I cannot reproduce the error.  Maybe because I have a different hardware set-up.
I also see that you are not running the latest version....

Edzelf, 2024-04-16:
Please, try the 06-MAR-2024 first.

thb584Git, 2024-04-17:
Hm, probably my bad. I see one release (5.5.22) and 5 tagged versions also latest 5.5.22 and the current project state. How to try 06-MAR-2024? Downloading zip of current project was what I did before (first post no version number in log file)
However if it's that what you mean I will give it another try in a clean environnement. Maybe something went wrong.

Edzelf, 2024-04-17:
Do not use the archive, download from the code page.  The version is at line 111 of main.cpp.

Edzelf, 2024-04-22:
I have made a set-up like yours with a PCM5102 module.  I still cannot reproduce the bug, but I will leave the radio on for a few days.  However there may be a problem with closing the stream when changing station.  The function stop_mp3client in main.cpp should be changed like this:
```
void stop_mp3client ()
{
  queueToPt ( QSTOPSONG ) ;                        // Queue a request to stop the song
  while ( mp3client && mp3client->connected() )    // Client active and connected?
  {
    ESP_LOGI ( TAG, "Stopping client" ) ;          // Yes, stop connection to host
    mp3client->close() ;                           // Close connection
    mp3client->abort() ;                           // Necesary to prevent crash
    vTaskDelay ( 500 / portTICK_PERIOD_MS ) ;
  }
}
```
This has been corrected in the latest version.

thb584Git, 2024-04-23:
Thank you for investigating, even not able to reproduce it. Well on normal use I saw two reboots the last days, both pretty soon after starting the radio tuning the station.
Still have no glue, I think the platform version is not the reason. The second device was in a state I had continous reboots not working at all. Persisting bahaviour after serveral times of compiling and fashing. Got back to work again after using your Arduino .ino code programming the file system. Might there be some inizialising missing causing random data accessed? However erase before upload as I apply should provide the same content.
I don't see the modified version on githb , I will patch the code and give it a try. Thanks