Edzelf/ESP32Radio-V2 #150: Display lights up but no data displayed

chainsawdave, 2024-07-04:
Love this project Ed. The back light from the display is on. I'm using your PCB. The backlight seems to be directly connected to the power supply. I'm using the 1.8 SPI 128 x 160
Here is the log:-
 *  Executing task in folder ESP32Radio-V2-main: C:\Users\mrdav\.platformio\penv\Scripts\platformio.exe device monitor --port COM11 

--- forcing DTR inactive
--- forcing RTS inactive

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Terminal on COM11 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[    33][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3067][I][main.cpp:2448] setup(): [main] Version Wed, 06 Mar 2024 16:00:00 GMT.  Free memory 110580
[  3069][I][main.cpp:2449] setup(): [main] Display type is BLUETFT
[  3157][I][main.cpp:2459] setup(): [main] SPIFFS is okay, space 956561, used 32881
[  3161][I][main.cpp:2480] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3163][I][main.cpp:2480] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3173][I][main.cpp:2480] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1048576
[  3182][I][main.cpp:2480] setup(): [main] Found partition 'coredump' at offset 0x3F0000 with size    65536
[  3193][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::($)pin_i2s_bck type=33
[  3199][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::($)pin_i2s_din type=33
[  3207][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::($)pin_i2s_lck type=33
[  3214][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::batt0 type=33
[  3221][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::batt100 type=33
[  3228][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::clk_dst type=33
[  3234][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::clk_offset type=33
[  3242][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::clk_server type=33
[  3249][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::gpio_00 type=33
[  3256][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::gpio_16 type=33
[  3262][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::gpio_17 type=33
[  3269][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::gpio_21 type=33
[  3276][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::gpio_25 type=33
[  3283][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::gpio_33 type=33
[  3290][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::ir_40BF type=33
[  3297][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::ir_C03F type=33
[  3304][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::mqttbroker type=33
[  3311][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::mqttpasswd type=33
[  3318][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::mqttport type=33
[  3325][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::mqttprefix type=33
[  3332][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::mqttuser type=33
[  3339][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::pin_enc_clk type=33
[  3346][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::pin_enc_dt type=33
[  3353][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::pin_enc_sw type=33
[  3360][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::pin_ir type=33
[  3367][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::pin_tft_cs type=33
[  3374][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::pin_tft_dc type=33
[  3381][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::pin_vs_cs type=33
[  3388][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::pin_vs_dcs type=33
[  3395][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::pin_vs_dreq type=33
[  3403][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::preset_00 type=33
[  3410][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::preset_01 type=33
[  3417][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::preset_02 type=33
[  3424][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::preset_03 type=33
[  3431][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::preset_04 type=33
[  3438][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::preset_05 type=33
[  3445][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::preset_06 type=33
[  3452][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::preset_07 type=33
[  3459][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::preset_08 type=33
[  3466][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::preset_09 type=33
[  3473][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::preset_10 type=33
[  3480][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::preset_11 type=33
[  3487][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::toneha type=33
[  3494][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::tonehf type=33
[  3500][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::tonela type=33
[  3507][I][main.cpp:2338] fillkeylist(): [main] ESP32-Radio::tonelf type=33
[... 103 lines cut ...]
[ 14885][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Expires: Thu, 01 Dec 2003 16:00:00 GMT
[ 14889][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Cache-Control: no-cache, must-revalidate
[ 14897][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Pragma: no-cache
[ 14904][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Access-Control-Allow-Header: *
[ 14913][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Access-Control-Allow-Methods: GET, HEAD
[ 14922][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Access-Control-Allow-Origin: *
[ 14930][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Access-Control-Allow-Credentials: true
[ 14939][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Access-Control-Expose-Headers: *
[ 14947][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Access-Control-Max-Age: 8640000
[ 14956][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: X-STW-lsid: cookie:e07d2f75-70ee-4589-a674-ea71289501a7
[ 14966][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Set-Cookie: uuid=e07d2f75-70ee-4589-a674-ea71289501a7; expires=Sat, 03 Aug 2024 22:18:54 GMT; path=/; domain=.live.streamtheworld.com
[ 14984][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Set-Cookie: uuid-s=e07d2f75-70ee-4589-a674-ea71289501a7; expires=Sat, 03 Aug 2024 22:18:54 GMT; path=/; domain=.live.streamtheworld.com; SameSite=None; Secure; HttpOnly
[ 15004][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Content-Type: audio/mpeg
[ 15011][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Connection: close
[ 15018][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: icy-name: Yacht Rock Hits
[ 15026][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: icy-description: The Biggest Hits Of The Yacht Rock Genre | 24/7/365
[ 15038][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: icy-url: http://spacial.com
[ 15046][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: icy-genre: 
[ 15052][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: icy-br: 128
[ 15059][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: icy-metaint: 16000
[ 15066][I][main.cpp:3555] handlebyte_ch(): [main] Headerline: Server: MediaGateway 6.7.3-0832.el6
[ 15075][I][main.cpp:3625] handlebyte_ch(): [main] Switch to DATA, bitrate is 128 kbps, metaint is 16000
[ 15084][I][main.cpp:4320] playtask(): [main] QSTARTSONG
[ 15196][I][main.cpp:1224] showstreamtitle(): [main] Streamtitle found, 44 bytes
[ 15196][I][main.cpp:1225] showstreamtitle(): [main] StreamTitle='Turley Richards - Stand By Me';

Edzelf, 2024-07-05:
Looks like the settings for pin_tft_cs en pin_tft_dc are reversed.  My definitieo in the config page of the web interface is lake this:
```
pin_tft_cs = 2                                       # GPIO Pin number for TFT "CS"
pin_tft_dc = 4                                       # GPIO Pin number for TFT "DC"
```

chainsawdave, 2024-07-05:
Thanks Ed will give this a go 😀Sent from my iPhone

chainsawdave, 2024-07-05:
Hi Ed Changed the values and after a few reboots, I resoldered the pins and not it works great. I've a build in mind for it. Will send pictures when complete. Thank you for all of your help. This is such an awesome project I've learned so much!
Cheers
David
[image]