Edzelf/ESP32Radio-V2 #152: https streams: how to play them?

MarcEngrie, 2024-09-09:
I have an https stream I want to hear: "https://streams.radio.dpgmedia.cloud/redirect/joe_gold/mp3".

Adding "streams.radio.dpgmedia.cloud/redirect/joe_gold/mp3" or "https://streams.radio.dpgmedia.cloud/redirect/joe_gold/mp3" into the config/preferences  does not work. 
Is the code capable of handling https streams? If so, how to let it know it is a https stream.

Thx

Edzelf, 2024-09-09:
Sorry, the radio cannot play SSL streams.
I do not understand why a station decides to use an encrypted stream.  Maybe the are affraid that someone listens in....

MarcEngrie, 2024-09-09:
Ha! OK. Thx!

ArminHobbyProjects, 2024-09-12:
Hi, 
I'm all new here, so I am reluctant to open a new topic ;-)
I also have a one stream address that is not playing: ais-sa8.cdnstream1.com/2749_64.aac
The track info is showing, but no sound. Works in browser and on a Frontier Silicon based webradio (Teufel360).
Any hints?

Thank you for the great work!!!
Armin

fanfanlatulipe26, 2024-09-12:
This stream is playing with my radio (running an old version  Th 03 Nov 2022 ...)
Tested with the web interface.
I pasted  ais-sa8.cdnstream1.com/2749_64.aac in the field "Enter a station/file here ..." and clicked PLAY

Edzelf, 2024-09-12:
Also tested it.  Plays well with my set-up (SPDIFF output).
If you are using a VS1003, AAC is not supported.

ArminHobbyProjects, 2024-09-12:
I tested it on a fresh build/download last night, using the Helix SPDIF optical output on a "china" ESP32-wroom board....
Also pasting the URL via the web interface

ArminHobbyProjects, 2024-09-16:
...just for documentation on how to obtain a 44,1k bitrate from a 22,05k stream:
In helixfuncs.h/playChunk(), double the samplerate and output each sample twice.

Though the 64kbit stream does not really sound adequate on HIFI equipment ;-(

Again thanks for the great effort&prompt support!
Armin