Edzelf/ESP32Radio-V2 #159: Help plese:-(

DanDimnet, 2024-12-02:
hello, I need advice please, I want to use only LCD, VS1053 and ESP32 without rotator, without infrared, without SD card. What should I comment on, please, and what will be the wiring diagram. thank you very much

Edzelf, 2024-12-03:
```
// ESP32dev Signal  Wired to LCD        Wired to VS1053    
// -------- ------  --------------      -------------------
// GPIO32           -                   pin 1 XDCS         
// GPIO33           -                   -                  
// GPIO5            -                   pin 2 XCS          
// GPIO4            pin 3 D/C or A0     -                  
// GPIO2            pin 2 CS            -                  
// GPIO16   RXD2    -                   -                  
// GPIO17   TXD2    -                   -                  
// GPIO18   SCK     pin 5 CLK or SCK    pin 5 SCK          
// GPIO19   MISO    -                   pin 7 MISO         
// GPIO23   MOSI    pin 4 DIN or SDA    pin 6 MOSI         
// GPIO15           -                   pin 4 DREQ         
// GPIO3    RXD0    -                   -                  
// GPIO1    TXD0    -                   -                  
// -------  ------  ---------------     -------------------
// GND      -       pin 8 GND           pin 8 GND          
// VCC 5 V  -       pin 7 BL            -                  
// VCC 5 V  -       pin 6 VCC           pin 9 5V           
// EN       -       pin 1 RST           pin 3 XRST  

In inlude/config.h: define (uncomment):
#define DEC_VS1053
#define BLUETFT
Uncomment and edit the definiton for FIXEDWIFI. 

Make sure the lines with “pin_tft_cs = 2” and “pin_tft_dc = 4” are defined:
pin_tft_cs = 2                                       # GPIO Pin number for TFT "CS"
pin_tft_dc = 4                                       # GPIO Pin number for TFT "DC"

As soon the webserver is running, edit the config page and define:
pin_vs_cs = 5
pin_vs_dcs = 32
pin_vs_dreq = 15

```

DanDimnet, 2024-12-03:
Thanks for the reply, there is no sound, and this is what I get on the serial monitor:

--- Terminal on COM5 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[    33][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[  3093][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[  3102][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[  3113][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3132][I][main.cpp:2499] setup(): [main] Version Thu, 17 Oct 2024 10:10:00 GMT.  Free memory 110580
[  3142][I][main.cpp:2500] setup(): [main] Display type is BLUETFT
[  3230][I][main.cpp:2510] setup(): [main] SPIFFS is okay, space 956561, used 33132
[  3240][I][main.cpp:2531] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3250][I][main.cpp:2531] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3260][I][main.cpp:2531] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1048576
[  3269][I][main.cpp:2531] setup(): [main] Found partition 'coredump' at offset 0x3F0000 with size    65536
[  3285][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_33 type=33
[  3292][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::ir_40BF type=33
[  3299][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::ir_C03F type=33
[  3306][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttbroker type=33
[  3313][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttpasswd type=33
[  3320][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttport type=33
[  3328][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttprefix type=33
[  3335][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttuser type=33
[  3342][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_enc_clk type=33
[  3349][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_enc_dt type=33
[  3357][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_enc_sw type=33
[  3364][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_ir type=33
[  3371][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_tft_cs type=33
[  3378][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_tft_dc type=33
[  3386][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_vs_cs type=33
[  3393][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_vs_dcs type=33
[  3400][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_vs_dreq type=33
[  3407][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset type=33
[  3414][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_00 type=33
[  3422][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_01 type=33
[  3429][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_02 type=33
[  3436][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_03 type=33
[  3443][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_04 type=33
[  3451][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_05 type=33
[  3458][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_06 type=33
[  3465][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_07 type=33
[  3473][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::batt0 type=33
[  3480][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::batt100 type=33
[  3487][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::clk_dst type=33
[  3494][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::clk_offset type=33
[  3501][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::clk_server type=33
[  3509][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_00 type=33
[  3516][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_16 type=33
[  3523][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_17 type=33
[  3530][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_21 type=33
[  3537][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_25 type=33
[  3544][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_08 type=33
[  3551][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_09 type=33
[  3558][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::toneha type=33
[  3565][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::tonehf type=33
[  3572][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::tonela type=33
[  3579][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::tonelf type=33
[  3586][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::volume type=33
[  3593][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::wifi_00 type=33
[  3600][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::wifi_01 type=33
[  3607][I][main.cpp:2402] fillkeylist(): [main] Read 45 keys from NVS
[  3620][I][main.cpp:1909] readIOprefs(): [main] 'pin_ir       ' set to 35
[  3628][I][main.cpp:1909] readIOprefs(): [main] 'pin_enc_clk  ' set to 12
[  3635][I][main.cpp:1909] readIOprefs(): [main] 'pin_enc_dt   ' set to 14
[  3643][I][main.cpp:1909] readIOprefs(): [main] 'pin_enc_sw   ' set to 13
[... cut ...]
[ 14757][I][main.cpp:3606] handlebyte_ch(): [main] Headerline: icy-url: http://www.radio1.nl
[ 14766][I][main.cpp:3606] handlebyte_ch(): [main] Headerline: icy-metaint: 16000
[ 14773][I][main.cpp:3676] handlebyte_ch(): [main] Switch to DATA, bitrate is 192 kbps, metaint is 16000
[ 14867][I][main.cpp:4371] playtask(): [main] QSTARTSONG
[ 15251][I][main.cpp:1291] showstreamtitle(): [main] Streamtitle found, 52 bytes
[ 15258][I][main.cpp:1292] showstreamtitle(): [main] StreamTitle='NPO Radio 1 - Spraakmakers - KRO-NCRV';

Edzelf, 2024-12-03:
Note this:
`[ 12770][W][VS1053.cpp:162] testComm(): [VS1053] This is not a VS1053, but a VS1003 instead!`
Define VS1003 in config.h instead of VS1053 or buy a real VS1053 board.

DanDimnet, 2024-12-03:
I edited the config.h on VS1003, see the picture, but it still gives me an error on the serial monitor

[image]

--- Terminal on COM5 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[  3093][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[  3102][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[  3113][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3133][I][main.cpp:2499] setup(): [main] Version Thu, 17 Oct 2024 10:10:00 GMT.  Free memory 110580
[  3142][I][main.cpp:2500] setup(): [main] Display type is BLUETFT
[  3230][I][main.cpp:2510] setup(): [main] SPIFFS is okay, space 956561, used 33132
[  3241][I][main.cpp:2531] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3251][I][main.cpp:2531] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3260][I][main.cpp:2531] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1048576
[  3270][I][main.cpp:2531] setup(): [main] Found partition 'coredump' at offset 0x3F0000 with size    65536
[  3283][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_00 type=33
[  3291][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_25 type=33
[  3298][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_33 type=33
[  3305][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_16 type=33
[  3312][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_17 type=33
[  3319][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_21 type=33
[  3326][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::batt100 type=33
[  3333][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::batt0 type=33
[  3340][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttbroker type=33
[  3347][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttport type=33
[  3355][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttuser type=33
[  3362][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttpasswd type=33
[  3369][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttprefix type=33
[  3377][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::wifi_00 type=33
[  3384][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::wifi_01 type=33
[  3391][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::volume type=33
[  3398][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::toneha type=33
[  3405][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::tonehf type=33
[  3412][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::tonela type=33
[  3419][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::tonelf type=33
[  3426][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset type=33
[  3434][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_00 type=33
[  3441][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_01 type=33
[  3448][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_02 type=33
[  3455][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_03 type=33
[  3463][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_04 type=33
[  3470][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_05 type=33
[  3477][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_06 type=33
[  3485][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_07 type=33
[  3492][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_08 type=33
[  3499][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_09 type=33
[  3506][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::clk_server type=33
[  3514][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::clk_offset type=33
[  3521][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::clk_dst type=33
[  3528][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::ir_40BF type=33
[  3535][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::ir_C03F type=33
[  3542][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_ir type=33
[  3549][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_enc_clk type=33
[  3557][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_enc_dt type=33
[  3564][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_enc_sw type=33
[  3572][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_tft_cs type=33
[  3579][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_tft_dc type=33
[  3586][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_vs_cs type=33
[  3593][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_vs_dcs type=33
[  3601][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_vs_dreq type=33
[  3608][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::($)pin_i2s_bck type=33
[  3616][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::($)pin_i2s_lck type=33
[  3624][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::($)pin_i2s_din type=33
[  3633][I][main.cpp:2402] fillkeylist(): [main] Read 48 keys from NVS
[  3648][I][main.cpp:1909] readIOprefs(): [main] 'pin_ir       ' set to 35
[... 141 lines cut ...]
[... cut ...]
[ 18577][I][main.cpp:3676] handlebyte_ch(): [main] Switch to DATA, bitrate is 192 kbps, metaint is 16000
[ 18577][I][main.cpp:3676] handlebyte_ch(): [main] Switch to DATA, bitrate is 192 kbps, metaint is 16000
[ 18587][I][main.cpp:4371] playtask(): [main] QSTARTSONG
[ 18587][I][main.cpp:4371] playtask(): [main] QSTARTSONG
[ 18726][I][main.cpp:1291] showstreamtitle(): [main] Streamtitle found, 91 bytes
[ 18733][I][main.cpp:1292] showstreamtitle(): [main] StreamTitle='NPO Radio 1 - Bureau Buitenland - Marine Le Pen laat haar tanden zien - VPRO';

Edzelf, 2024-12-03:
```
[ 15510][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:0384 R1:0380 R2:0384
[ 15561][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:35E8 R1:35E8 R2:35E0
[ 15649][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:972C R1:972C R2:9760
[ 15730][E][VS1053.cpp:150] testComm(): [VS1053] SPI error. SB:F168 R1:F168 R2:C600
```
There is still something wrong with the SPI connection to the VS1003 board or the board itself is faulty.

DanDimnet, 2024-12-03:
thank you, I will try it, I will order a new module, now I have a PCM5102 module, but I don't know how to connect it and what to define in config.h and on the web interface. thank you for your willingness

DanDimnet, 2024-12-07:
Hello, please help, I connected the PCM5102 module and the pins according to the configuration. It doesn't make any sound, I only have power connected and the three pins LCK, DIN, BCK I send the dump.

--- Terminal on COM5 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    33][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[  3100][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[  3109][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[  3120][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3139][I][main.cpp:2499] setup(): [main] Version Thu, 17 Oct 2024 10:10:00 GMT.  Free memory 110580
[  3149][I][main.cpp:2500] setup(): [main] Display type is BLUETFT
[  3239][I][main.cpp:2510] setup(): [main] SPIFFS is okay, space 956561, used 33132
[  3250][I][main.cpp:2531] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3260][I][main.cpp:2531] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3270][I][main.cpp:2531] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1048576
[  3279][I][main.cpp:2531] setup(): [main] Found partition 'coredump' at offset 0x3F0000 with size    65536
[  3293][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::($)pin_i2s_bck type=33
[  3301][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::($)pin_i2s_din type=33
[  3309][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::($)pin_i2s_lck type=33
[  3316][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::batt0 type=33
[  3323][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::batt100 type=33
[  3330][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::clk_dst type=33
[  3337][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::clk_offset type=33
[  3345][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::clk_server type=33
[  3352][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_00 type=33
[  3359][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_16 type=33
[  3366][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_17 type=33
[  3373][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_21 type=33
[  3381][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_25 type=33
[  3388][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::gpio_33 type=33
[  3395][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::ir_40BF type=33
[  3402][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::ir_C03F type=33
[  3409][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttbroker type=33
[  3416][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttpasswd type=33
[  3424][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttport type=33
[  3431][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttprefix type=33
[  3438][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::mqttuser type=33
[  3445][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_enc_clk type=33
[  3453][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_enc_dt type=33
[  3460][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_enc_sw type=33
[  3467][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_ir type=33
[  3474][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_tft_cs type=33
[  3482][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_tft_dc type=33
[  3489][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_vs_cs type=33
[  3496][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_vs_dcs type=33
[  3504][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::pin_vs_dreq type=33
[  3511][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_00 type=33
[  3518][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_01 type=33
[  3526][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_02 type=33
[  3533][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_03 type=33
[  3540][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_04 type=33
[  3550][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_05 type=33
[  3558][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_06 type=33
[  3565][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_07 type=33
[  3572][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_08 type=33
[  3579][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_09 type=33
[  3587][I][main.cpp:2389] fillkeylist(): [main] ESP32-Radio::preset_10 type=33
[... 140 lines cut ...]
[ 16374][I][helixfuncs.h:297] playChunk(): [helixfuncs] Channels    is 2
[ 16380][I][helixfuncs.h:298] playChunk(): [helixfuncs] Bitpersamp  is 16
[ 16387][I][helixfuncs.h:299] playChunk(): [helixfuncs] Outputsamps is 2304
[ 22983][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[... cut ...]
[358333][I][main.cpp:1291] showstreamtitle(): [main] Streamtitle found, 54 bytes
[358341][I][main.cpp:1292] showstreamtitle(): [main] StreamTitle='Signál Rádio - To nejlepší z 80.let';
[442893][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[503026][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
[506849][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[511330][I][main.cpp:1291] showstreamtitle(): [main] Streamtitle found, 51 bytes
[511338][I][main.cpp:1292] showstreamtitle(): [main] StreamTitle='KYLIE MINOGUE - I SHOULD BE SO LUCKY';

[image]

Edzelf, 2024-12-07:
Defian in the configuration web page (example, define your pins):
```
pin_i2s_bck = 15                                     # GPIO Pin number for I2S "BCK"
pin_i2s_din = 26                                     # GPIO Pin number for I2S "DIN"
pin_i2s_lck = 27                                     # GPIO Pin number for I2S "L(R)CK"
```

DanDimnet, 2024-12-07:
Yes, I have it connected like that, I also tried two PCM5102 modules and another ESP32 and nothing

[image]

Edzelf, 2024-12-07:
Have you read [this](https://github.com/Edzelf/ESP32Radio-V2/blob/main/doc/ESP32-radio%20buiding%20instructions.pdf)?  See the important note at paragraph 2.2.

DanDimnet, 2024-12-07:
[image]

Edzelf, 2024-12-07:
SCK should be connected to GND.

DanDimnet, 2024-12-07:
hi i did that too, i even tried high or low level from this site https://arduino.stackexchange.com/questions/96102/no-output-from-pcm5102-i2s-dac and still no sound :-(

Edzelf, 2024-12-09:
Maybe this helps.  Connect the pin “3” on the back side of the module to “H”, pin “4” to “L”, pin “1” to “L” and pin 2 to “L”.  See the picture:
[image]

DanDimnet, 2024-12-09:
hi, it works for me now, but I'll go back to VS1003/VS1053. Thank you so much for the support anyway. I still wanted to ask, I have a fast connection to the Internet, ping 6.3, download 110Mbps, but the problem is already 192kbps, and the problem is that when downloading a file and listening to the radio, the sound of the radio is interrupted. As if there is little memory to save the radio stream. Could it be strengthened somehow to store more data?

Edzelf, 2024-12-10:
You may try to add extra bufferspace by increasing the definition of QSIZ in the main program, but I don''t think this will help much.  Note that the system will be unstable if there is less free memory.
It looks more like a problem in the WiFi network.

DanDimnet, 2024-12-12:
And the use of a better ESP32 that I have more memory? I still have the first version compiled from ESP8266, could it be modified there too? Is there something like SPIRAM? Sorry for my English

trip5, 2024-12-13:
Sorry to interrupt, especially if it is nothing... but I also get choppy playback (using a VS1053) anything above 128kbps, especially if the radio is too far from my wireless router and if there is a wall, it won't even do 128 for longer than 2 minutes.  So, I've been working on radios that will do longer distances...

I've been using ESP32-S3 with antennas (but ESP32 should be fine too)... like this: 
[image]
https://www.aliexpress.com/item/1005007235089505.html

With this:
[image]
https://www.aliexpress.com/item/32901995764.html

It works pretty well now but still can't do above 128kpbs.  I'm really hoping with the S3 (with it's various extra memory), there's some way to fix this... but I also haven't played with the PCM module yet... btw your picture has `($)` in front of the pin definitions.  I don't think those are supposed to be there.

It should be:
```
pin_i2s_bck = 15                                     # GPIO Pin number for I2S "BCK"
pin_i2s_din = 26                                     # GPIO Pin number for I2S "DIN"
pin_i2s_lck = 27                                     # GPIO Pin number for I2S "L(R)CK"
```
as Edzelf has already said.

DanDimnet, 2024-12-16:
Hi Ed, what did you use to make the board and schematic?  Is it possible to ask for the source data?  And maybe send to email dimnet@dimnet.cz?  And regarding my question about SPIRAM or SRAM for longer storage for the buffer is it possible?  And one more question, is the SD card functional for MP3 playback?  Thank you very much for your willingness and I appreciate your time to answer.

Edzelf, 2024-12-17:
I( used EasyEDA for the board and schematic design.  See the doc directory for the schematic.
I do not think SRAM will cure your WifI network.  SD card support is experimental.

DanDimnet, 2024-12-18:
Hi, I tried importing online to easydata. The first screen of import settings and the second screen of errors. I tried both Professional and Standart imports
[image]

[image]

DanDimnet, 2024-12-20:
Can you please check whether the zip file is not damaged or why it cannot be imported?  thank you very much

DanDimnet, 2025-01-25:
Hello, can you please advise me how to open the schematic and edit the PCB? EasyData won't open it for me. thank you very much

DanDimnet, 2025-01-25:
I need to edit your design for a small modification and I would like to ask you to send the data for import into EasyEDA so that I can edit the schematic and the board. Thank you very much for your willingness, my email is dimnet@dimnet.cz