Edzelf/ESP32Radio-V2 #161: ESP32-ETH01

darkaragon, 2024-12-08:
Good morning,
i want to build a radio based on your project and need to use eth. First of all: im not very experienced with arduino or programming code yet, so i need to learn.
I keep ethernet working with 1106 oled display (i2c pin TXD and 458_EN)) but im not shure where to connect the audio interface PCM5102A with i2s and the rotary encoder.
Audio Interface of course is necessary for the radio.
Can you help me with this issue? Many thanks!

Edzelf, 2024-12-08:
You need 3 free pins for the PCM5102A.  Define DEC_HELIX in include/config.h and the 3 pins in the config page of the web interface:
```
pin_i2s_bck = 15                                     # GPIO Pin number for I2S "BCK"
pin_i2s_din = 26                                     # GPIO Pin number for I2S "DIN"
pin_i2s_lck = 27                                     # GPIO Pin number for I2S "L(R)CK"
```
Note the the pin-numbers mentioned above are just an example.
Read the "Important note" at chapter 2.2 of [this](https://github.com/Edzelf/ESP32Radio-V2/blob/main/doc/ESP32-radio%20buiding%20instructions.pdf) document and connect the CLK pin of the PCM5102A module to GND.

darkaragon, 2024-12-09:
Many thanks for your quick response, i keep it working by setting DEC_HELIX (not recognized the i2s comment)
so with wt32-eth01 i use

pin_enc_clk = 39     # GPIO rotary encoder CLK
pin_enc_dt = 35      # GPIO rotary encoder DT
pin_enc_sw = 4      # GPIO rotary encoder SW
pin_i2s_bck = 14    # GPIO Pin number for I2S "BCK"/ PCM5102
pin_i2s_din = 12    # GPIO Pin number for I2S "DIN"/ PCM5102
pin_i2s_lck = 15     # GPIO Pin number for I2S "L(R)CK"/ PCM5102
pin_tft_scl = 33      # GPIO Pin number for SCL / 1106 Display
pin_tft_sda = 17     # GPIO Pin number for SDA / 1106 Display

and its working fine. Many thanks for your work. One last question: is there any configuration possibility for Mono mixdown output?

greetings, claas

Edzelf, 2024-12-10:
There is no setting for mono mixdown.  You may change helixfuncs.h, but you may also mix the 2 channels on the analog output.