Edzelf/ESP32Radio-V2 #162: Read 0 keys from NVS

61pete, 2024-12-15:
Im running Arduino IDE 1.8.19 as per page 17 of the pdf.
SPIFFS is V2.0.0 and using the esp32fs.jar tool to upload the data folder in the SPIFFS format.

**Which version of ESP32 by espressif should I be using?**

V2.0.0 fails to compile at 2739: 'ADC_ATTEN_DB_12' was not declared in this scope

V2.0.10 fails at line 126:unused variable 'vstype' [-Werror=unused-variable] 

V2.0.15 compiles and uploads, but NVS fails to read pins

V2.0.17 compiles and uploads, but NVS fails to read pins

V3.0.0 fails at line 2384:too few arguments to function 'esp_err_t nvs_entry_find(const char*, const char*, nvs_type_t, nvs_opaque_iterator_t**)'

V3.0.7 fails at line 2384:too few arguments to function 'esp_err_t nvs_entry_find(const char*, const char*, nvs_type_t, nvs_opaque_iterator_t**)'

This is the serial monitor with V2.0.17:

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3018][I][ESP32Radio.ino:2499] setup(): [main] Version Thu, 17 Oct 2024 10:10:00 GMT.  Free memory 110580
[  3028][I][ESP32Radio.ino:2500] setup(): [main] Display type is OLED
[  3183][I][ESP32Radio.ino:2510] setup(): [main] SPIFFS is okay, space 1318001, used 0
[  3276][I][ESP32Radio.ino:2531] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3286][I][ESP32Radio.ino:2531] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3296][I][ESP32Radio.ino:2531] setup(): [main] Found partition 'spiffs  ' at offset 0x290000 with size  1441792
[  3306][I][ESP32Radio.ino:2531] setup(): [main] Found partition 'coredump' at offset 0x3F0000 with size    65536
[  3317][I][ESP32Radio.ino:2402] fillkeylist(): [main] Read 0 keys from NVS
[  3325][I][ESP32Radio.ino:1909] readIOprefs(): [main] 'pin_spi_sck  ' set to 18
[  3332][I][ESP32Radio.ino:1909] readIOprefs(): [main] 'pin_spi_miso ' set to 19
[  3339][I][ESP32Radio.ino:1909] readIOprefs(): [main] 'pin_spi_mosi ' set to 23
[  3356][I][ESP32Radio.ino:2570] setup(): [main] GPIO0 is HIGH
[  3371][I][ESP32Radio.ino:2570] setup(): [main] GPIO2 is HIGH

61pete, 2024-12-15:
I tried my hand with Platformio and got the same result.

[env:esp32dev]
platform = espressif32 @ ^6.9.0
board = esp32dev
framework = arduino
board_build.partitions = radiopart.csv
upload_port = COM13

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3028][I][main.cpp:2499] setup(): [main] Version Thu, 17 Oct 2024 10:10:00 GMT.  Free memory 110580
[  3037][I][main.cpp:2500] setup(): [main] Display type is OLED
[  3125][I][main.cpp:2510] setup(): [main] SPIFFS is okay, space 956561, used 33132
[  3135][I][main.cpp:2531] setup(): [main] Found partition 'nvs     ' at offset 0x009000 with size    20480
[  3145][I][main.cpp:2531] setup(): [main] Found partition 'otadata ' at offset 0x00E000 with size     8192
[  3155][I][main.cpp:2531] setup(): [main] Found partition 'spiffs  ' at offset 0x2F0000 with size  1048576
[  3164][I][main.cpp:2531] setup(): [main] Found partition 'coredump' at offset 0x3F0000 with size    65536
[  3176][I][main.cpp:2402] fillkeylist(): [main] **Read 0 keys from NVS**
[  3183][I][main.cpp:1909] readIOprefs(): [main] 'pin_spi_sck  ' set to 18
[  3189][I][main.cpp:1909] readIOprefs(): [main] 'pin_spi_miso ' set to 19
[  3196][I][main.cpp:1909] readIOprefs(): [main] 'pin_spi_mosi ' set to 23
[  3213][I][main.cpp:2570] setup(): [main] GPIO0 is HIGH

Edzelf, 2024-12-15:
Use the web interface to load the NVS. See chapter 2.2 and 11 in the document.

61pete, 2024-12-16:
Thanks, that worked for me.

closed 2024-12-16 by 61pete