Edzelf/ESP32Radio-V2 #163: Compilation errors

lumtech, 2024-12-19:
Two computers, and the same errors:
Arduino IDE 2.3.4 and 1.8.19
ESP32lib - 3.1,0
ESP32 Dev Module
All actions on compiling files and libraries were done according to the instructions: ESP32-radio buiding instructions.pdf
ESP32-Radio.pdf.

Maybe Im using a raw version of ESP IDF? (3,1,0)
Although in Config.h it is written: - // Needs platform = espressif32@3.1.0 !!!!

C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\aac_decoder.cpp: In function 'int DecodeOneSymbol(int)':
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\aac_decoder.cpp:8534:57: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
 8534 |     nBits = DecodeHuffmanScalar(huffTabSBR, hi, bitBuf, &val);
      |                                                         ^~~~
      |                                                         |
      |                                                         int*
In file included from C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\aac_decoder.cpp:10:
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\aac_decoder.h:487:115: note:   initializing argument 4 of 'int DecodeHuffmanScalar(const short int*, const HuffInfo_t*, unsigned int, int32_t*)'
  487 | int DecodeHuffmanScalar(const signed short *huffTab, const HuffInfo_t *huffTabInfo, unsigned int bitBuf, int32_t *val);
      |                                                                                                          ~~~~~~~~~^~~
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino: In function 'void otastart()':
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino:1654:3: error: 'timerAlarmDisable' was not declared in this scope
 1654 |   timerAlarmDisable ( timer ) ;                    // Disable the timer
      |   ^~~~~~~~~~~~~~~~~
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino: In function 'void fillkeylist()':
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino:2384:23: error: too few arguments to function 'esp_err_t nvs_entry_find(const char*, const char*, nvs_type_t, nvs_opaque_iterator_t**)'
 2384 |   it = nvs_entry_find ( "nvs", NAME, NVS_TYPE_ANY ) ;           // Get first entry
      |        ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino:121:
C:\Users\LENOVO\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-083aad99-v2\esp32/include/nvs_flash/include/nvs.h:716:11: note: declared here
  716 | esp_err_t nvs_entry_find(const char *part_name,
      |           ^~~~~~~~~~~~~~
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino:2398:27: error: cannot convert 'nvs_iterator_t' {aka 'nvs_opaque_iterator_t*'} to 'nvs_opaque_iterator_t**'
 2398 |     it = nvs_entry_next ( it ) ;
      |                           ^~
      |                           |
      |                           nvs_iterator_t {aka nvs_opaque_iterator_t*}
C:\Users\LENOVO\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-083aad99-v2\esp32/include/nvs_flash/include/nvs.h:775:42: note:   initializing argument 1 of 'esp_err_t nvs_entry_next(nvs_opaque_iterator_t**)'
  775 | esp_err_t nvs_entry_next(nvs_iterator_t *iterator);
      |                          ~~~~~~~~~~~~~~~~^~~~~~~~
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino: In function 'void setup()':
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino:2639:32: error: 'TCPIP_ADAPTER_IF_STA' was not declared in this scope
 2639 |   tcpip_adapter_set_hostname ( TCPIP_ADAPTER_IF_STA,
      |                                ^~~~~~~~~~~~~~~~~~~~
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino:2639:3: error: 'tcpip_adapter_set_hostname' was not declared in this scope
 2639 |   tcpip_adapter_set_hostname ( TCPIP_ADAPTER_IF_STA,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino:2693:22: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
 2693 |   timer = timerBegin ( 0, 80, true ) ;                   // User 1st timer with prescaler 80
      |           ~~~~~~~~~~~^~~~~~~~~~~~~~~
In file included from C:\Users\LENOVO\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0\cores\esp32/esp32-hal.h:98,
                 from C:\Users\LENOVO\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0\cores\esp32/Arduino.h:36,
                 from C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino:115:
C:\Users\LENOVO\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino:2694:24: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
 2694 |   timerAttachInterrupt ( timer, &timer100, false ) ;     // Call timer100() on timer alarm
      |   ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\LENOVO\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino:2695:3: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
 2695 |   timerAlarmWrite ( timer, 100000, true ) ;              // Alarm every 100 msec
      |   ^~~~~~~~~~~~~~~
      |   timerWrite
C:\Users\LENOVO\Documents\Arduino\ESP32Radio-V2\ESP32Radio\ESP32Radio.ino:2696:3: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
 2696 |   timerAlarmEnable ( timer ) ;                           // Enable the timer
      |   ^~~~~~~~~~~~~~~~
      |   timerAlarm
[... 48 lines cut ...]
  //#define DEC_HELIX_AI                                    // Software decoder for AI Audio kit (AC101)
  #define DEC_HELIX_INT                                   // Software decoder for MP3, AAC. DAC output
                                                            // Needs platform = espressif32@3.1.0 !!!!

  // Define (just one) type of display.  See documentation.
 // #define BLUETFT                                         // Works also for RED TFT 128x160
  //#define ST7789                                          // 240x240 TFT (SPI)
  //#define OLED1306                                        // 64x128 I2C OLED SSD1306
  //#define OLED1309                                        // 64x128 I2C OLED SSD1309
  //#define OLED1106                                        // 64x128 I2C OLED SH1106
  #define DUMMYTFT                                          // Dummy display
  //#define LCD1602I2C                                      // LCD 1602 display with I2C backpack
  //#define LCD2004I2C                                      // LCD 2004 display with I2C backpack
  //#define ILI9341                                         // ILI9341 240*320
  //#define NEXTION                                         // Nextion display
  //

  // Define ZIPPYB5 if a ZIPPY B5 Side Switch is used instead of a rotary switch
  //#define ZIPPYB5

  // End of configuration parameters.
  #define CONFIG_H
#endif

//////////////////////////////////////////////////////////////////////////////////////////////

aly-fly, 2024-12-19:
**Leave the Arduino behind and move to PlatformIO.** 
Reserve 2-3 hours each day over the weekend, watch some YT tutorials and make some experiments. It's a relatively involving step-up to familiarize yourself with more exposed environment, but once you make it, you will never look back to Arduino.

Edzelf, 2024-12-19:
Yes, it is much easier than to struggle with the Arduino IDE.

aly-fly, 2024-12-29:
I use PlatformIO. From the .ini file I deleted all version references to old libraries and frameworks. It compiles and works fine with the most recent versions. Even DAC output for which is written in the code that it should have an ancient version of the framework. Obviously the issues were fixed.

dasflaigsi, 2025-01-22:
Espressif has a guid for the breaking changes of the arduino esp32 core v3

https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html