Edzelf/ESP32Radio-V2 #169: Stop command is missing

AndrewLinden, 2025-01-03:
Hello!
In v1 there was a stop command, now there is no stop command.
When making a radio without physical buttons and screen, to be controlled only over network with MQTT, it is very useful to stop the stream and leave the radio idle waiting for next time it is needed.

PS! When implementing stop command also check it will shutdown amp if a shutdown pin is defined (and using I2S).

trip5, 2025-01-03:
I got around this by sending an mqtt command for it to play a file that didn't exist (ie from a stream or file).

aly-fly, 2025-01-03:
Open Main.cpp
Find function `const char* analyzeCmd ( const char* par, const char* val )`
Find line `  else if ( argument == "status" )                    // Status request`
Above it add the following section:

```
  else if ( argument == "stop" )                       // Stop request?
  {
    myQueueSend ( sdqueue, &stopcmd ) ;                // Stop player SD card
    myQueueSend ( radioqueue, &stopcmd ) ;             // Stop player web radio
    //also clear display
    tftset ( 1, "" ) ;                                 // Clear artist
    tftset ( 2, "  STOP" ) ;                           // Clear song 
    tftset ( 3, "" ) ;                                 // Clear station
// add here also mqtt messages if needed
  }
  else if ( argument == "start" )                     // Start request?
  {
    myQueueSend ( radioqueue, &startcmd  ) ;          // Restart radio player
  }
  else if ( argument == "sleep" )                     // Sleep request?
  {
    sleepreq = true ;                                 // Yes, set request flag
  }
```

AndrewLinden, 2025-01-12:
Thank you, @aly-fly, it works!