Edzelf/ESP32Radio-V2 #171: Problem Building and Uploading in PlatformIO

omlott, 2025-01-25:
I realize it has todo something with the "ESPAsycTCP" Library? The one that is included via PIO Home is called "AsycTCP" Following is the terminal output of the failing upload

Processing esp32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing me-no-dev/ESPAsyncWebServer @ 1.2.3
Unpacking  [####################################]  100%
Library Manager: ESP Async WebServer@1.2.3 has been installed!
Library Manager: Resolving dependencies...
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (6.9.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain+, Compatibility ~ soft
Library Manager: Installing me-no-dev/ESPAsyncWebServer @ 1.2.3
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: ESP Async WebServer@1.2.3 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing Hash
Found 57 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Wire @ 2.0.0
|-- PubSubClient @ 2.8.0
|-- Adafruit BusIO @ 1.17.0
|-- Adafruit GFX Library @ 1.11.11
|-- Adafruit ILI9341 @ 1.6.1
|-- Adafruit ST7735 and ST7789 Library @ 1.10.4
|-- AC101 @ 0.0.1
|-- Ch376msc @ 1.4.5
|-- AsyncTCP @ 1.1.1
|-- WiFi @ 2.0.0
|-- ESP Async WebServer @ 1.2.3
|-- FS @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- SPI @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- codecs
|-- LCD1602
Building in release mode
Archiving .pio/build/esp32/lib58d/libAsyncTCP.a
Indexing .pio/build/esp32/lib58d/libAsyncTCP.a
Archiving .pio/build/esp32/lib813/libWiFi.a
Indexing .pio/build/esp32/lib813/libWiFi.a
Compiling .pio/build/esp32/libfa8/ESPAsyncTCP/AsyncPrinter.cpp.o
Compiling .pio/build/esp32/libfa8/ESPAsyncTCP/ESPAsyncTCP.cpp.o
Compiling .pio/build/esp32/libfa8/ESPAsyncTCP/ESPAsyncTCPbuffer.cpp.o
Compiling .pio/build/esp32/libfa8/ESPAsyncTCP/SyncClient.cpp.o
Archiving .pio/build/esp32/libd98/libFS.a
Indexing .pio/build/esp32/libd98/libFS.a
In file included from /home/omletta/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/opt.h:51,
                 from /home/omletta/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/init.h:40,
                 from .pio/libdeps/esp32/ESPAsyncTCP/src/ESPAsyncTCP.h:31,
                 from .pio/libdeps/esp32/ESPAsyncTCP/src/AsyncPrinter.h:26,
                 from .pio/libdeps/esp32/ESPAsyncTCP/src/AsyncPrinter.cpp:22:
/home/omletta/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/port/esp32/include/lwipopts.h:439: warning: "TCP_MSS" redefined
 #define TCP_MSS                         CONFIG_LWIP_TCP_MSS
 
In file included from .pio/libdeps/esp32/ESPAsyncTCP/src/ESPAsyncTCP.h:25,
                 from .pio/libdeps/esp32/ESPAsyncTCP/src/AsyncPrinter.h:26,
                 from .pio/libdeps/esp32/ESPAsyncTCP/src/AsyncPrinter.cpp:22:
.pio/libdeps/esp32/ESPAsyncTCP/src/async_config.h:13: note: this is the location of the previous definition
[... 213 lines cut ...]
In file included from /home/omletta/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:192,
                 from .pio/libdeps/esp32/ESPAsyncTCP/src/ESPAsyncTCP.cpp:74:
/home/omletta/.platformio/packages/framework-arduinoespressif32/cores/esp32/IPAddress.h:50:24: note:   initializing argument 1 of 'IPAddress::IPAddress(uint32_t)'
     IPAddress(uint32_t address);
               ~~~~~~~~~^~~~~~~
.pio/libdeps/esp32/ESPAsyncTCP/src/ESPAsyncTCP.cpp: In member function 'void AsyncServer::begin()':
.pio/libdeps/esp32/ESPAsyncTCP/src/ESPAsyncTCP.cpp:1128:23: error: invalid user-defined conversion from 'IPAddress' to 'const ip_addr_t*' {aka 'const ip_addr*'} [-fpermissive]
   err = tcp_bind(pcb, local_addr, _port);
                       ^~~~~~~~~~
In file included from /home/omletta/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:192,
                 from .pio/libdeps/esp32/ESPAsyncTCP/src/ESPAsyncTCP.cpp:74:
/home/omletta/.platformio/packages/framework-arduinoespressif32/cores/esp32/IPAddress.h:59:5: note: candidate is: 'IPAddress::operator uint32_t() const' <near match>
     operator uint32_t() const
     ^~~~~~~~
/home/omletta/.platformio/packages/framework-arduinoespressif32/cores/esp32/IPAddress.h:59:5: note:   no known conversion from 'uint32_t' {aka 'unsigned int'} to 'const ip_addr_t*' {aka 'const ip_addr*'}
In file included from .pio/libdeps/esp32/ESPAsyncTCP/src/ESPAsyncTCP.cpp:79:
/home/omletta/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/tcp.h:458:18: note:   initializing argument 2 of 'err_t tcp_bind(tcp_pcb*, const ip_addr_t*, u16_t)'
 err_t            tcp_bind    (struct tcp_pcb *pcb, const ip_addr_t *ipaddr,
                  ^~~~~~~~
*** [.pio/build/esp32/libfa8/ESPAsyncTCP/AsyncPrinter.cpp.o] Error 1
*** [.pio/build/esp32/libfa8/ESPAsyncTCP/ESPAsyncTCP.cpp.o] Error 1
========================================================================= [FAILED] Took 5.39 seconds =========================================================================

 *  The terminal process "platformio 'run', '--target', 'upload', '--upload-port', '/dev/ttyUSB0'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

chainsawdave, 2025-01-31:
I'm not much good at his coding stuff, eager to learn. I think I'm having the same issue. It just started when I want compile a new ESP32 with the code.
Cannot find "C:/Users/mrdav/OneDrive/Documents/PlatformIO/Projects/ESP32Radio-V2-main (3)/ESP32Radio-V2-main/.pio/libdeps/esp32/ESP Async WebServer/src"

It looks like the github was moved in January 2025. I don't know how to modify this part of the code.

Edzelf, 2025-02-03:
It seems that the me-no-dev version of the async web server has diappeared.  Change the lib_deps part of the platformio.ini file to:
```
lib_deps = 
  Wire
  knolleary/PubSubClient@^2.8
  adafruit/Adafruit BusIO@^1.9.3
  adafruit/Adafruit GFX Library@^1.10.4
  adafruit/Adafruit ILI9341@^1.5.9
  adafruit/Adafruit ST7735 and ST7789 Library@^1.7.5
  mathieucarbou/ESP Async WebServer @ ^3.0.6
  yveaux/AC101@^0.0.1
  djuseeq/Ch376msc @ ^1.4.4
```

omlott, 2025-02-03:
Thanks, you got it. Now building and uploaded is working directly. When connecting to the AP after first boot 192.168.4.1 is just a blank screen saying "sorry". But that also might be a problem for a new issue or is it connected to the altered version of web server now?

[image]

Edzelf, 2025-02-03:
"Sorry" is displayed if you have forgotten to upload the web-interface from the "data" directory.

Ryszard-wpl, 2025-02-07:
What am I doing wrong - cannot compile :
".....
successfully created esp32 image.
Compiling .pio/build/esp32/lib6c7/Wire/Wire.cpp.o
Compiling .pio/build/esp32/lib428/PubSubClient/PubSubClient.cpp.o
In file included from /Users/rw/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:29,
                 from /Users/rw/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:33,
                 from src/main.cpp:119:
/Users/rw/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:39:10: fatal error: Network.h: No such file or directory

*****************************************************************
* Looking for Network.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Network.h"
* Web  > https://registry.platformio.org/search?q=header:Network.h
*
*****************************************************************

   39 | #include "Network.h"
      |          ^~~~~~~~~~~
compilation terminated.
Compiling .pio/build/esp32/libb28/SPI/SPI.cpp.o
*** [.pio/build/esp32/src/main.cpp.o] Error 1
=============================================== [FAILED] Took 3.51 seconds ==============================================="

is it problem with libraries or can I install Network.h file by manualy?

chainsawdave, 2025-02-09:
Hi 👋 I’m no expert on this. However, the library changed January this year. Ed sent some code you can copy into the platform.io file. That might help. I’ve had compiling issues but the think I have some duff boards. Will try with some new one. Good luck.Cheers DavidSent from my iPhone

mrburns42, 2025-02-28:
I am seeing the same build problem.    If I ask Visual Studio Code to go to the declaration of Network.h,  then the file exists as shown below.     I did a make clean.   I also stopped and restarted Visual Studio Code.

Not sure how to debug this since it is an espressif header file that PlatformIO downloaded.    

**********************************************************************
/*
 * SPDX-FileCopyrightText: 2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */
#pragma once

#include "NetworkInterface.h"
#include "NetworkEvents.h"
#include "NetworkManager.h"

#include "NetworkClient.h"
#include "NetworkServer.h"
#include "NetworkUdp.h"

mrburns42, 2025-02-28:
Since the file actually existed,  I guessed that it was a stale configuration within the platformio  files.    I closed Visual Studio Code, then I went down into the User folder shown below and deleted the two folders dealing with espressif.    Then I restarted Visual Studio Code and allowed it to download and reconfigure the espressif stuff.    Once all downloaded,  (takes a while).   Then the Build completed perfectly without issue.  

 ("smill" is my username below, so you will have to substitute your name when going down into the User folder.) 

C:\Users\smill\.platformio\platforms

Full disclosure,  I downloaded and attempted to build V1 of the ESP32Radio first before I discovered the later version.    It could have been the failed compiles of V1 which hosed my system.