Edzelf/ESP32Radio-V2 #175: 3.3v relay module

lubiezunia, 2025-02-26:
Hi

I need to turn on / off 3.3v relay using IR remote ( for power on / off external amplifier ).

I was looking for more information about pin_shutdown but without success. Can it be used for my purpose?
Any advice ?

Best regards
Mateusz

Edzelf, 2025-02-27:
The pin you define by "pin_shutdown" or "pin_shutdownx" will be toggled on play/stop.
It is documented in the pdf file.

lubiezunia, 2025-02-27:
so assuming that in config file I have:

pin_shutdown = 16  (or  pin_shutdownx = 16)  signal pin for my relay module - which should change state it's from HIGH to LOW or vice versa ?

what IR command I should use to trigger  play / stop ? doc. says "Amplifier shutdown pin. Pin will be set if the radio is not playing or the volume is set to zero. This output-pin can be used to shut down the amplifier." In may case volume = 0 or mute command doesn't do anything. Serial monitor should also report changed state of my pin 16?

Edzelf, 2025-02-27:
What are you using for decoder (like DEC_VS1053, EC_HELIX_SPDIF)?
It could be that some decoder does not handle the shutdown pin.

lubiezunia, 2025-02-27:
It plays well with 3ohm Samsung speakers but I need to figure out how to turn off my 200W amplifier when I'm not using the radio.

[image]

Edzelf, 2025-02-27:
So you are using the VS1053.  Then the pin(s) should change when volume changes from or to zero.

lubiezunia, 2025-02-27:
Ok my fault  - I should check it earlier - HIGH level goes LOW on multimeter on mute / (un)mute commands. 

So problem is on relay module side... 

Thanks Edzelf for your time

btw. if anyone is looking for complete transformers module for noise reduction I can say that I tested HCW-M171 and it works like a charm.

[image]

lubiezunia, 2025-02-27:
Relay module was faulty - second one works as expected.

lubiezunia, 2025-02-27:
Hi - my relay module turns OFF for 1 second every time I change radio channel by using downpreset, uppreset or preset. Is it normal?