Edzelf/ESP32Radio-V2 #180: Playback does not start or stops working randomly

friebi, 2025-03-15:
I have a minimal setup consisting of [ESP32-S3-DevKitC-1](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitc-1/user_guide.html#hardware-reference) and a VS1053. No display, no rotary or whatsoever.

I have multiple issues with the playback.

### No sound after boot
Even if everything works fine after boot, I have to modify the volume at least once in order to have sound output. Anyone else facing this issue?

### Stopping sound output or no sound output at all
After a random time (from few seconds to many (60+) minutes) the sound output just stops without any message on the console. The stream seems to continue, I also receive MQTT messages with title updates.

Sometimes I have the issue already directly after boot.

<details>
<summary>My Config</summary>

```
clk_dst = 2                                          # Offset during daylight saving time (hours)
clk_offset = 1                                       # Offset with respect to UTC in hours
clk_server = pool.ntp.org                            # Time server to be used
#
mqttbroker = mymqttserver                            # mqtt server
mqttpasswd = ************                            # mqtt password
mqttport = 1883                                      # mqtt port
mqttprefix = ESP32Radio                              # mqtt topic
mqttuser = mqtt                                      # mqtt user
#
pin_vs_cs = 5                                        # GPIO Pin number for VS1053 "CS"
pin_vs_dcs = 15                                      # GPIO Pin number for VS1053 "DCS"
pin_vs_dreq = 4                                      # GPIO Pin number for VS1053 "DREQ"
#
preset = 0
preset_00 = dispatcher.rndfnk.com/hr/hr1/mittelhessen/mp3/128/stream.mp3 #  0 - HR1 Mittelhessen
preset_01 = dispatcher.rndfnk.com/br/br1/schwaben/mp3/128/stream.mp3     #  1 - Bayern 1 Schwaben
preset_02 = dispatcher.rndfnk.com/dlf/03/mp3/128/stream.mp3              #  2 - DLF Nova
preset_03 = orf-live.ors-shoutcast.at/fm4-q1a                            #  3 - FM4
#
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
volume = 100
#
wifi_00 = SSID2/pw2
```

</details>

Additionally I use the SPI pins of ESP32-S3-DevKitC-1
- MOSI --> 11
- MISO --> 13
- SCK --> 12

<details>
<summary>Log example</summary>

```
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[  3233][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(44) txPin(43)
[  3242][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[  3251][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.

Starting ESP32-radio running on CPU 1 at 240 MHz.
[  3262][I][main.cpp:2499] setup(): [main] Version Thu, 17 Oct 2024 10:10:00 GMT.  Free memory 286708
[  3271][I][main.cpp:2500] setup(): [main] Display type is DUMMY
[... 161 lines cut ...]
[ 17337][E][main.cpp:3557] handlebyte_ch(): [main] MP3 packet dropped!
[ 17543][E][main.cpp:3557] handlebyte_ch(): [main] MP3 packet dropped!
[ 17713][I][main.cpp:4365] playtask(): [main] QSTARTSONG
[ 17766][I][main.cpp:659] publishtopic(): [main] Publish to topic ESP32Radio/playing : 1
[ 17774][I][main.cpp:1291] showstreamtitle(): [main] Streamtitle found, 59 bytes
[ 17781][I][main.cpp:1292] showstreamtitle(): [main] StreamTitle='hr1 Hörertelefon: 069 155 1111 - www.hr1.de';
[ 17862][I][main.cpp:659] publishtopic(): [main] Publish to topic ESP32Radio/icy/streamtitle : hr1 Hörertelefon: 069 155 1111 - www.hr1.de
[ 17871][D][AsyncTCP.cpp:173] _get_async_event(): coalescing polls, network congestion or async callbacks might be too slow!
[ 17885][D][AsyncTCP.cpp:173] _get_async_event(): coalescing polls, network congestion or async callbacks might be too slow!
[ 21373][I][main.cpp:659] publishtopic(): [main] Publish to topic ESP32Radio/volume : 98
[ 22165][I][main.cpp:659] publishtopic(): [main] Publish to topic ESP32Radio/volume : 100
[ 23894][I][main.cpp:1291] showstreamtitle(): [main] Streamtitle found, 52 bytes
[ 23901][I][main.cpp:1292] showstreamtitle(): [main] StreamTitle='Just around the corner von Cock Robin';
```

</details>

Things I've tried
- modifying volume (like mentioned above)
- restarting the stream or switching stations
- trying different GPIOs for the VS1053 connection

Only after a reset I am able to get sound output again.

I'd be happy for any idea 😄

Edzelf, 2025-03-19:
I have not tested the ESP-S3 very well.
The volume setting should be saved every 10 minutes in NVS by a call to handleSaveReq() in the main loop.  So after a reboot it should be set to the saved value.
The message "MP3 packet dropped" means that the sender sends data faster than the VS1053 can handle.  Maybe the clockfrequency of the VS1053 is a little off.

Edzelf, 2025-03-19:
Check the logging.  You should see a line with "volume = 100".

friebi, 2025-03-19:
Thanks for your answer. I already attached a log to my initial post. Here the excerpt:
```
[ 16254][I][main.cpp:3670] handlebyte_ch(): [main] Switch to DATA, bitrate is 128 kbps, metaint is 16384
[ 16263][I][main.cpp:659] publishtopic(): [main] Publish to topic ESP32Radio/icy/name : hr hr1 mittelhessen
[ 16366][I][main.cpp:659] publishtopic(): [main] Publish to topic ESP32Radio/preset : 0
[ 16466][I][main.cpp:659] publishtopic(): [main] Publish to topic ESP32Radio/volume : 100
[ 16513][E][main.cpp:3557] handlebyte_ch(): [main] MP3 packet dropped!
[ 16719][E][main.cpp:3557] handlebyte_ch(): [main] MP3 packet dropped!
[ 16925][E][main.cpp:3557] handlebyte_ch(): [main] MP3 packet dropped!
[ 17131][E][main.cpp:3557] handlebyte_ch(): [main] MP3 packet dropped!
[ 17337][E][main.cpp:3557] handlebyte_ch(): [main] MP3 packet dropped!
[ 17543][E][main.cpp:3557] handlebyte_ch(): [main] MP3 packet dropped!
[ 17713][I][main.cpp:4365] playtask(): [main] QSTARTSONG
[ 17766][I][main.cpp:659] publishtopic(): [main] Publish to topic ESP32Radio/playing : 1
[ 17774][I][main.cpp:1291] showstreamtitle(): [main] Streamtitle found, 59 bytes
[ 17781][I][main.cpp:1292] showstreamtitle(): [main] StreamTitle='hr1 Hörertelefon: 069 155 1111 - www.hr1.de';
[ 17862][I][main.cpp:659] publishtopic(): [main] Publish to topic ESP32Radio/icy/streamtitle : hr1 Hörertelefon: 069 155 1111 - www.hr1.de
[ 17871][D][AsyncTCP.cpp:173] _get_async_event(): coalescing polls, network congestion or async callbacks might be too slow!
[ 17885][D][AsyncTCP.cpp:173] _get_async_event(): coalescing polls, network congestion or async callbacks might be too slow!
[ 21373][I][main.cpp:659] publishtopic(): [main] Publish to topic ESP32Radio/volume : 98
[ 22165][I][main.cpp:659] publishtopic(): [main] Publish to topic ESP32Radio/volume : 100
```
You see in line 4 volume is set to 100, but I have no sound. In the last but one line you see I reduced the volume manually to 98. This is the moment where I start to have sound output.

Edzelf, 2025-03-19:
I saw that the saved volume is not in the logging.  Check if you can see in in the config page of the web interface. Like this:

```
#
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
volume = 80
#
```

friebi, 2025-03-20:
The pasted config in the OP is copied from the web interface. It has
```
#
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
volume = 100
```