Edzelf/ESP32Radio-V2 #181: Audio (S/PDIF) gaps when accessing the webinterface

FOTO-Miro, 2025-03-19:
Hello Es,
First of all, congratulations for your great work.

I've now completely switched to digital.

Current configuration:

ESP32 + Display + S/PDIF, nothing more.

So far, everything's perfect.

Every time I access the web interface or interact with it, there are gaps in the sound. It's as if every action briefly interrupts the sound.

Is there anything known about this, or do you have any ideas?
THANK YOU

Edzelf, 2025-03-19:
The processor is heavily loaded by the decoding software.  The web interface also needs some CPU capacity.  So short gaps are normal.

FOTO-Miro, 2025-03-20:
OK. Thank you.
Is external decoder with SPDIF Output possible ?