Edzelf/ESP32Radio-V2 #182: found some bugs in the cbmp3list function???

xP373Rx, 2025-04-21:
in my copilation  current version the sd track list was screwed up first track was double last track was not there and 1 \n was missing after the first track.

Here is the fixed code  that works for me.
`size_t cb_mp3list ( uint8_t *buffer, size_t maxLen, size_t index )
{
  static int         i ;                              // Index in track list
  static const char* path ;                           // Pointer in file path
  size_t             len = 0 ;                        // Number of bytes filled in buffer
  char*              p = (char*)buffer ;              // Treat as pointer to aray of char
  static bool        eolSeen ;                        // Remember if End Of List

  if ( index == 0 )                                   // First call for this page?
  {
    i = 0 ;                                           // Yes, set index (track number)
    path = getFirstSDFileName() ;                     // Force read of next path
    eolSeen = ( path == nullptr ) ;                   // Any file?
  }
  while ( ( maxLen > len ) && ( ! eolSeen ) )         // Space for another char from path?
  {
    
    
    if ( *path )                                      // End of path?
    {
      *p++ = *path++ ;                                // No, add another character to send buffer
      len++ ;                                         // Update total length
    }
    
    else
    {
      // End of path
      if ( i< SD_filecount-1)                                        // At least one path in output?
      {
        *p++ = '\n' ;                                 // Yes, add separator
        len++ ;                                       // Update total length
      }

      i++;
      path = getSDFileName ( i ) ;                  // Get next path from list
      if ( i >= SD_filecount )                        // No more files?
      {
        eolSeen = true ;                              // Yes, stop
        break ;
      }
    }
  }
  // We arrive here if output buffer is completely full or end of tracklist is reached
  return len ;                                        // Return filled length of buffer
}`

Pako2, 2025-04-22:
I solved the same problem a year and a half ago: [#125](https://github.com/Edzelf/ESP32Radio-V2/issues/125)
Unfortunately, the author didn't respond at all. That was one of the reasons why I ended up abandoning this project completely and going my [own way](https://github.com/Pako2/RadioESP32) (although I used a lot of stuff from here).

xP373Rx, 2025-04-22:
Well dude has probably no time to maintain ^^ life can be a b1tch ;>