Edzelf/ESP32Radio-V2 #184: error: 'I2S_COMM_FORMAT_STAND_I2MSB' was not declared in this scope

krpagar, 2025-06-28:
Hello Sir, I would say its fantastic job! I was looking for something similiar and finally got it. However while copiling and facing some issue like below as pasted. Could you please help resolve this? 

src/main.cpp: In function 'void playtask(void*)':
src/main.cpp:4449:41: error: 'I2S_COMM_FORMAT_STAND_I2MSB' was not declared in this scope
       i2s_config.communication_format = I2S_COMM_FORMAT_STAND_I2MSB ;  // Yes, use new definition
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:4449:41: note: suggested alternative: 'I2S_COMM_FORMAT_STAND_MSB'
       i2s_config.communication_format = I2S_COMM_FORMAT_STAND_I2MSB ;  // Yes, use new definition
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                                         I2S_COMM_FORMAT_STAND_MSB

Edzelf, 2025-06-29:
This error occurs if "DEC_HELIX" is defined in config.h.  The solution is to follow the compiler suggestion.
So change 'I2S_COMM_FORMAT_STAND_I2MSB' into 'I2S_COMM_FORMAT_STAND_MSB'.  I will correct the sourcecode.

krpagar, 2025-06-29:
Yes. Its working finally! thank you for your time in responding.